adafruit/ArduinoCore-samd #6: Variant header pin numbers not correct.

aryelevin, 2017-08-31:
Hi,

The PIN numbers of SPI, SPI1 and USB is not correct as in the variant.c file.
Also, the I2C and SPI1 both using sercom5.

Another issue, in the SPI, you use PORTA 14 for the MOSI pin, and PA14 is SERCOM4 PAD[2] which require you to use SPI_PAD_2_SCK_3 which is the only option for using MOSI as PAD[2], and SCK needs to be at PAD[3] which already taken by the MISO pin which is PA15 serves as PAD[3].
I just switched to use PA13 as MOSI, and I'm using SPI_PAD_0_SCK_1 and everything is fine.

Thanks for the awesome work on this.
Arye

aryelevin, 2017-09-01:
In addition to that, PA12 is used as D12 pin as well as SCK.
I switched to use SCK in PB13 which is same sercom4[1].

deanm1278, 2017-09-20:
thanks! I have fixed SPI and the USB pin numbers for the hardware we are using. Not sure currently what pins SPI1 will be on in our final design.
https://github.com/adafruit/ArduinoCore-samd/commit/c0c9ba0058960ab598936934512dd93114a44fc9

closed 2017-09-20 by deanm1278