adafruit/ArduinoCore-samd #7: No floating point

WestfW, 2017-09-14:
A big advantage of the D51 is hardware floating point, but I don't see any attempts to turn it on, yet, either in the init code (it might be automatic in Atmel startup files?) or in the compiler options.   This is a little fraught with peril, as an immediate effect of floating point is an additional 128 bytes of registers that have to be preserved during context switching (ie for ISRs.)  CM4F has a feature called "Lazy context save of FP state" that should probably be used.  With "lazy", stack space is reserved for the FP registers in ISRs, but not actually saved unless you use a floating-point instruction.

(Do you want "issues" reported against D51 yet?  Or should I go away until you say "go"?)

ladyada, 2017-09-15:
hiya, yep please just open issues, with info and (best of all) instructions on how to reproduce or fix. chips are still not available so we have not done more work on this platform yet

WestfW, 2017-09-16:
TI Launchpad (Energia) uses compile options:
compiler.ldflags=-mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant

deanm1278, 2017-09-20:
thanks! FPU support has been added https://github.com/adafruit/ArduinoCore-samd/commit/13707df6f0d67929bbd0d724873c1f37446db76c

The only way I have currently been able to verify it's operation is that running the following code with FPU enabled gives a performance boost (from 7436ms to 7184ms):

`Serial.begin(9600);
  while(!Serial);
  Serial.println("FPU test");
  
  unsigned long t = millis();

  float accum = 0;
  for( unsigned int i = 0; i < 10000000; ++i )
    {
        accum += (float)( rand( ) % 365 );
    }

    Serial.println(millis() - t);`

closed 2017-09-20 by deanm1278