adafruit/ArduinoCore-samd #10: SAMD51 Bootloader

MartinL1, 2017-09-26:
Hi @deanm1278 

I'm currently unable to used the Arduino IDE to upload the SAMD51 bootloader. 

My SAMD51J20A hardware is working and I'm able to sucessfully load GCC C Executable Projects using Atmel Studio 7 and have got the microcontroller clocking at 120MHz, however I'm unable to load the SAMD51 bootloader using the Arduino IDE. On attempting to burn the SAMD51 bootloader with the Arduino IDE I get the follwing errors:

`...Error: SAMD: NVM Error
Error: SAMD: NVM lock error
Error: SAMD: NVM programming error
Error: Failed to erase row containing 00804000
Error: Couldn't erase user row...`

Simply loading the bootloader file "...metro4/samdx1_sam_ba.bin" using Atmel Studio, causes the USB to enumerate as an Arduino Zero in bootloader mode, but this only happens from a reset condition (by pressing the reset button), not from a cold start. After the successful compilation of a blank sketch, the Arduino IDE fails to upload, with the message:

`No device found on COM7
An error occurred while uploading the sketch`

Also, the Arduino IDE is unable to program the sketch directly using my Atmel-ICE programmer directly, as it appears that it doesn't recognise the SAMD51's device ID.

Please could you tell me if there's something obvious that I'm missing?

MartinL1, 2017-09-26:
Despite being able to run GCC C Executable Projects, I just noticed that my SAMD51 is failing to read the NVM Software Calibration registers in Atmel Studio, (Tools->Device Programming->Fuses Tab). Could be the reason why it's not working as expected? 

I'm going to make a second SAMD51 board as a comparison.

MartinL1, 2017-09-26:
Further to my last comment, I soldered second SAMD51 board. 

It behaves the same as the first board, runs GCC C Executable projects in Atmel Studio just fine, but unable to upload the bootloader via the Arduino IDE. (Fuse settings identical to the first board).

deanm1278, 2017-09-26:
hey @MartinL1, there is currently a bug causing USB to not enumerate immediately when plugged in. It should enumerate after a reset condition though. This will hopefully be fixed as soon as I can track it down (USB stack is complicated).

For the Arduino IDE, you will need a modified version of BOSSA that supports the SAMD51 devices found here https://github.com/adafruit/BOSSA/tree/arduino (make sure you build the arduino branch)

That goes in place of the existing bossac executable Arduino uses (on windows this is C:\Users\yourusername\AppData\Local\arduino15\packages\arduino\tools\bossac\1.7.0\bossac.exe )

MartinL1, 2017-09-26:
Hi @deanm1278 

Thanks for you response.

Yes, I previously placed the modified BOSSA executable in the Arduino's 1.7.0 directory in anticipation on uploading sketches. 

My current issue though is with the SAMD51 bootloader. I'm unable burn the bootloader using the Arduino IDE and Atmel-ICE, as OpenOCD says there's a NVM Lock Error. Do you know if there are any fuses I need to set beforehand?

Uploading the bootloader (...metroM4/samdx1_sam_ba.bin) using Atmel Studio, causes the USB to enumerate, but upon reset as you suggest. On the Arduino IDE the toolchain compiles and links just fine, however for me the upload timesout with error "No device found on COM7, an error occured while uploading the sketch". 

I'm not sure what I'm doing wrong?

deanm1278, 2017-09-26:
But you can't upload sketches with the arduino ide after flashing the bootloader with atmel studio?

Does it work if you program using bossac directly from the command line?

MartinL1, 2017-09-27:
Hi @deanm1278 

> But you can't upload sketches with the arduino ide after flashing the bootloader with atmel studio?

Yes, that's right.

> Does it work if you program using bossac directly from the command line?

I tried using bossac on the command line with the modified bossac.exe file:

`bossac.exe --port=COM7 -U true -i -e -w -v -b C:\Users\Computer\Docmuments\Arduino\Blink\Blink.ino.metro_m4.bin -R`

...but it returns the message "No device found on COM7". 

The thing is, that after plugging the USB and pressing the reset button, Windows detects the board and I can see it in the Windows device manager as "Arduino Zero Bootloader (COM7)". 

Also, I tested the bossac command line using the same modified bossac.exe file for a Blink sketch on my SAMD21G18A board and that worked perfectly.

MartinL1, 2017-09-27:
In order to test that I'm using the correct procedure, I went through the same process on my SAMD21G18A board. I burned the Arduino Zero bootloader (samd21_sam_ba.bin) on to the SAMD21 using Atmel Studio, then successfully uploaded a sketch using the Arduino IDE.

deanm1278, 2017-09-27:
hey @MartinL1 can you try using this bossac executable? Make sure the board has enumerated properly and run:
`bossac.exe -d -i`

and then copy paste any output here. thanks!

[bossac.zip](https://github.com/adafruit/ArduinoCore-samd/files/1337525/bossac.zip)

MartinL1, 2017-09-27:
Hi @deanm1278 

Yes, that works! Thank you.

deanm1278, 2017-09-27:
ok excellent @MartinL1,
That's the bossac from the repo so there must have just been a problem when replacing it.
If you have it all set up with the arduino IDE now I'm gonna go ahead and close the issue.
Please let me know of any bugs you come across while using it :)

closed 2017-09-27 by deanm1278