adafruit/ArduinoCore-samd #11: I2C Bus BAUD Register Overflow

MartinL1, 2017-10-03:
Hi @deanm1278 

Just a couple of little things.

In the "SERCOM.cpp" file on line 444:

`sercom->I2CM.BAUD.bit.BAUD = SystemCoreClock / ( 2 * baudrate) - 1 ;`

The high SystemCoreClock value of 120,000,000 is causing the 8-bit BAUD register overflow at low "baudrate" values. This affects I2C in standard mode at 100kHz, where it causes the bus to clock at around 735kHz. Fast mode at 400kHz remains unaffected however.

The only solution I can see at the moment is to run the I2C peripheral off the 48MHz DFLL clock source instead, like the SAMD21.

Also, in the "startup.c" file on line 167:

` OSCCTRL->Dpll[1].DPLLRATIO.reg = OSCCTRL_DPLLRATIO_LDRFRAC(0x00) | OSCCTRL_DPLLRATIO_LDR(0x49); //100 Mhz`

I think that the OSCCTRL_DPLLRATIO_LDR should be set to 49 decimal.

I'm also having some timing issues with Serial1. If I set the baud rate to 115200 with Serial.begin(115200), serial works with some errors, but only at a baud rate of around 57600. I'm currently trying to track down the source the problem.

deanm1278, 2017-10-03:
Hey make sure you've pulled the most recent version of the core, I fixed
some sercom baud rate stuff last week.

MartinL1, 2017-10-04:
Hi @deanm1278 

I found the isse with the Serial1, it's in the "SERCOM.cpp" file, line 736:

`GCLK->PCHCTRL[clk_core].reg = GCLK_PCHCTRL_GEN_GCLK2_Val | (1 << GCLK_PCHCTRL_CHEN_Pos);`

This is connecting the SERCOM to the 100MHz clock source, however the SERCOM::initUART() function (at the top of the SERCOM.cpp file) is using the 120MHz (SystemCoreClock) to determine the baud rate.

Just changing the clock source to the 120MHz solves this:

`GCLK->PCHCTRL[clk_core].reg = GCLK_PCHCTRL_GEN_GCLK0_Val | (1 << GCLK_PCHCTRL_CHEN_Pos);`

deanm1278, 2017-10-04:
oh I guess I never pushed that commit last week: https://github.com/adafruit/ArduinoCore-samd/commit/e42ab7bd9083bcd8e145124c522a17b946a6adf6

sorry about that it's pushed now if you pull the master branch. The datasheet says the SERCOM max clock rate is 100Mhz so we probably don't want to clock it from the core 120Mhz clock

deanm1278, 2017-10-04:
@MartinL1 since I know you're using this core currently, I have made updates to the bootloader size. Please see here if you have issues:
https://github.com/adafruit/ArduinoCore-samd/issues/12

MartinL1, 2017-10-04:
Hi @deanm1278 

> The datasheet says the SERCOM max clock rate is 100Mhz so we probably don't want to clock it from the core 120Mhz clock

Oops, I missed that.

> since I know you're using this core currently, I have made updates to the bootloader size. Please see here if you have issues:

Many thanks. I'll update everything.

deanm1278, 2017-10-04:
@MartinL1 just looked into the DPLL1 issue and yea you're right it was supposed to be 49 decimal to generate the 100Mhz clock.
also looking at the I2C baud rate overflow, yea it appears that only 8 bits are available in the BAUD register so we cannot generate a 100khz SCL from 100Mhz source.
Because of this I'll leave the sercom clocked from the 48mhz dfll for now.
https://github.com/adafruit/ArduinoCore-samd/commit/fb41af46059d5e2308ec68644bd8a2b53a7aa9db

closed 2017-10-04 by deanm1278