adafruit/ArduinoCore-samd #13: No device driver for updated bootloader

MartinL1, 2017-10-05:
Hi @deanm1278, 

I tried to use the updated bootloader, but the Arduino IDE is unable to detect it, as there doesn't appear to be a device driver. I tried downloading the Adafruit device drivers for the Metro M0, but this didn't work.

For now I've gone back to the old one with the Arduino build VIDs and PIDs.

asprakash, 2017-10-11:
I am also facing the same issue. My Feather M0 board is not detected in bootloader mode once I use updated bootloader.

deanm1278, 2017-10-11:
@asprakash the updated bootloader was only for the samd51 M4 boards currently under development. Your feather M0 did not require any update. Please revert to whatever setup you had before.

deanm1278, 2017-10-11:
@MartinL1 if you program the attached bootloader onto your board, and pull the latest version of the samd51 branch your IDE does not detect the board?

[samd51_bootloader.zip](https://github.com/adafruit/ArduinoCore-samd/files/1377251/samd51_bootloader.zip)

it should show up as:
[image]

does the board show up in your system's device manager?

What OS are you running?

deanm1278, 2017-10-11:
ok @MartinL1 if you're using Windows 7 I guess you need a driver. @ladyada is going to update the driver shortly.

ladyada, 2017-10-11:
the INF is updated here https://github.com/adafruit/Adafruit_Windows_Drivers/blob/master/Drivers/Adafruit_usbser/Adafruit_usbser.inf its not signed at this time, so you'll need to force-install it

MartinL1, 2017-10-12:
Hi @deanm1278 

I pulled the latest version and tried the zipped bootloader, but the Arduino IDE still won't recognize the board. I'm using Window 8.1.

The situation boils down to this. If I burn the old "samdx1_bootloader.bin" file with the Arduino build VID/PID in the "boards.txt" and use the old 8K bootloader linker_script it works:

`adafruit_metro_m4.build.vid=0x2341
adafruit_metro_m4.build.pid=0x804d
adafruit_metro_m4.bootloader.tool=openocd
adafruit_metro_m4.bootloader.file=metroM4/samdx1_sam_ba.bin`

If on the other hand, I burn the new "samd51_bootloader" file and the Adafruit build VID/PID in the "boards.txt" file and the new 16K linker_script, it doesn't:

`adafruit_metro_m4.build.vid=0x239A
adafruit_metro_m4.build.pid=0x8020
adafruit_metro_m4.bootloader.tool=openocd
adafruit_metro_m4.bootloader.file=metroM4/samd51_sam_ba.bin`

Perhaps I've missed a step somewhere? Is there another file I need to change?

deanm1278, 2017-10-13:
hey @MartinL1 ok yea it looks like a driver issue. Did you try to force install the updated driver ladyada linked above?
It will eventually be signed before release, but until that happens force install will be necessary for your OS

MartinL1, 2017-10-17:
Hi @deanm1278 

> Did you try to force install the updated driver ladyada linked above?

I force installed the updated driver that ladyada linked above. On Windows, as the driver is currently unsigned I first had to disable device signature enforcement.

After burning the "samd51_sam_ba.bin" bootloader, changing the linker_script to account for the 16K and altering the "boards.txt" file with the Adafruit VID/PIDs, the board is only able to operate in bootloader mode. 

I'm finding that the USB won't re-emumerate after upload and the sketch doesn't start. Also the sketch doesn't run either from a single press of the reset button, or after removing then re-applying power.

deanm1278, 2017-10-18:
@MartinL1 things should work ok if you pull the latest version of samd51 branch and have all USB drivers installed.
There are 2 major bugs that popped up that I am unable to find at the moment (see the readme). Both have workarounds so the rest of the core should be functional.

deanm1278, 2017-10-18:
@MartinL1 actually just fixed the bug that caused the board to be in bootloader mode to upload sketches. Having the board in bootloader mode to upload sketches is no longer necessary with the most recent pull of the core.

MartinL1, 2017-10-19:
Hi @deanm1278 

In the latest pull of the core, there's no bootloader binary.

Also, building the latest bootloader using the "Makefile" in Atmel Studio 7 generates compilation errors, while building an earlier version of the bootloader compiles and generates the binary just fine.

deanm1278, 2017-10-19:
@MartinL1 you can use this bootloader:
[bootloader.zip](https://github.com/adafruit/ArduinoCore-samd/files/1398880/bootloader.zip)

What are the compilation errors you are getting? Note that with the new bootloader you will need to build either the 'metrom4' or the 'featherm4' configuration.

MartinL1, 2017-10-19:
Hi @deanm1278

> Note that with the new bootloader you will need to build either the 'metrom4' or the 'featherm4' configuration.

I'm using metrom4.

> What are the compilation errors you are getting?

It generates errors building the file "board_driver_serial.h":

`Severity	Code	Description	Project	File	Line
Warning		implicit declaration of function '_UI' [-Wimplicit-function-declaration]	samd51_sam_ba	C:\Users\Computer\AppData\Local\Arduino15\packages\arduino-beta\hardware\samd\1.0.0\bootloaders\metroM4\board_driver_serial.h	31

Error		enumerator value for 'UART_RX_PAD0_TX_PAD2' is not an integer constant	samd51_sam_ba	C:\Users\Computer\AppData\Local\Arduino15\packages\arduino-beta\hardware\samd\1.0.0\bootloaders\metroM4\board_driver_serial.h	31

Error		enumerator value for 'UART_RX_PAD1_TX_PAD2' is not an integer constant	samd51_sam_ba	C:\Users\Computer\AppData\Local\Arduino15\packages\arduino-beta\hardware\samd\1.0.0\bootloaders\metroM4\board_driver_serial.h	32

Error		enumerator value for 'UART_RX_PAD2_TX_PAD0' is not an integer constant	samd51_sam_ba	C:\Users\Computer\AppData\Local\Arduino15\packages\arduino-beta\hardware\samd\1.0.0\bootloaders\metroM4\board_driver_serial.h	33

Error		enumerator value for 'UART_RX_PAD3_TX_PAD0' is not an integer constant	samd51_sam_ba	C:\Users\Computer\AppData\Local\Arduino15\packages\arduino-beta\hardware\samd\1.0.0\bootloaders\metroM4\board_driver_serial.h	34

Error		enumerator value for 'UART_RX_PAD1_TX_PAD0' is not an integer constant	samd51_sam_ba	C:\Users\Computer\AppData\Local\Arduino15\packages\arduino-beta\hardware\samd\1.0.0\bootloaders\metroM4\board_driver_serial.h	35

Error		enumerator value for 'UART_RX_PAD3_TX_PAD2' is not an integer constant	samd51_sam_ba	C:\Users\Computer\AppData\Local\Arduino15\packages\arduino-beta\hardware\samd\1.0.0\bootloaders\metroM4\board_driver_serial.h	36

Error		recipe for target 'build/board_driver_serial.o' failed	samd51_sam_ba	C:\Users\Computer\AppData\Local\Arduino15\packages\arduino-beta\hardware\samd\1.0.0\bootloaders\metroM4\Makefile	126`

deanm1278, 2017-10-19:
@MartinL1 I can't seem to reproduce the issue, but it may be due to the fact that Adafruit has switched to the SAMD51J19A instead of the SAMD51J20A for the next rev of our metro M4 due to availability of chips. That is what we will be shipping on our hardware. The only differences are the flash and SRAM availability, but if your hardware is still using the SAMD51J20 you may want to fork the repo :/

MartinL1, 2017-10-20:
Hi @deanm1278 

Thanks for trying to reproduce the issue and for the bootloader zip file.

In the end, I managed to track down what was causing the problem. The move to the SAMD51J19A caused the compiler to pull in the "samd51j19a.h" file from the Atmel CMSIS. However, I hadn't updated this file to avoid the previous _U abd _L definition conflict with the ARM toolchain. Updating the "samd51j19a.h" file with the new definitions fixed the issue.

I've also made some progress in tracking down the issue regarding my SAMD51 board uploading OK in bootloader mode, but failing to start the sketch. It turns out that this problem isn't anything to do with drivers, but has something to do with the 16K bootloader size.

If I rebuild the "METRO_M4_samba.bin" bootloader with the "bootloader_METRO_M4.ld" linker script set to a bootloader size of 8K and use the old 8K "flash_with_bootloader.ld" file in the ".../variants/linker_scripts/..." directory, the board works normally and the USB re-enumerates after leaving bootloader mode. If however I use the 16K bootloader size in the "bootloader_METRO_M4.ld" and "flash_with_bootloader.ld" files, the board's USB fails to connect after exiting bootloader mode. It appears as though the bootloader is failing to correctly index the start address of the sketch.

I'm investigating, but haven't yet figured out what's causing it?

deanm1278, 2017-10-20:
Hey @MartinL1 if it works when using the old 8k bootloader size and linker script then that sounds like you're still using the old bossac. Try using this one and verify from the bossac output when arduino ide tries to upload that it says it is writing to 0x4000 instead of 0x2000
[bossac.zip](https://github.com/adafruit/ArduinoCore-samd/files/1402305/bossac.zip)

MartinL1, 2017-10-20:
Hi @deanm1278 

Yes, you're right, I was using an old "bossac.exe" file. Everything's now working as smooth as silk with the 16K linker scripts.

Thank you again for your support and guidance.

I'll close this issue, as it's been solved.

closed 2017-10-20 by MartinL1