adafruit/ArduinoCore-samd #15: MCK_VARIANT causing timing issues

MartinL1, 2017-10-12:
Hi @deanm1278 

In the process of debugging my SAMD51 board, I noticed that the MCK_VARIANT definition remains at 48000000ul. It's in the "variant.h" file, at line 33:

`#define VARIANT_MCK			  (48000000ul)`

However, VARIANT_MCK is used the SysTick calculations, in the file "delay.c" on line 59:

`return ((count+pend) * 1000) + (((SysTick->LOAD  - ticks)*(1048576/(VARIANT_MCK/1000000)))>>20) ;`

On the SAMD51 the SysTick->LOAD is set at 119999 rather than the 47999 on the SAMD21, so to maintain the correct timing for the delay(), micros() and millis() functions, VARIANT_MCK needs to be changed to 120000000ul.

MartinL1, 2017-10-12:
Hi @deanm1278 

I also made some small modifications to the "WInterrupts.c" file to account for the multiple NVIC EIC channels, in order to get the interrupts going. If you're interested, I've added it as an attachment "WInterrupts.zip"

The main difference is that on the SAMD51 (unlike the SAMD21), the EIC peripheral must be disabled in the CTRLA register, before writing to the CONFIG registers.

[WInterrupts.zip](https://github.com/adafruit/ArduinoCore-samd/files/1378719/WInterrupts.zip)

deanm1278, 2017-10-13:
@MartinL1 yea you're right VARIANT_MCK should be 120Mhz and I had forgotten about the EIC :)
EIC code you gave works and has been pushed and MCK has been updated
https://github.com/adafruit/ArduinoCore-samd/commit/63ba6aa24e72b894d4a12d38fdd34b57eec40228

thanks for the help!

deanm1278, 2017-10-13:
also @MartinL1 feel free to sumbit pull reqs next time you have some code as well

closed 2017-10-13 by deanm1278