adafruit/ArduinoCore-samd #16: latest SAMD boot loader requires manual entry to boot loader

jerryneedell, 2017-10-24:
The new Uf2 Bootloader [https://github.com/tannewt/uf2-samd](url) works for both Circuit python and Arduino but I found that with Arduino sketches, I had to manually enter the boot loader (double tap RESET) every time I wanted to upload a new sketch. Is this the intended operation.

Thanks for the fix. It is great not to have to change boot loaders and this is a workable issue, but should be made clear if it cannot be avoided.

ladyada, 2017-10-24:
we can probably fix it, @deanm1278 can you take a look? probably a magicnum thing?

deanm1278, 2017-10-24:
yup, I've fixed in this PR:
https://github.com/tannewt/uf2-samd/pull/2

ladyada, 2017-11-12:
reopen if still an issue!

closed 2017-11-12 by ladyada