adafruit/ArduinoCore-samd #21: not sure SERCOM2 will work for chosen SPI pins

dhalbert, 2017-11-16:
@deanm @ladyada It looks like you changed the primary SPI SERCOM for Metro M4 from `SERCOM4` to `SERCOM2` about a month ago, in 45f3eebae.
https://github.com/adafruit/ArduinoCore-samd/blob/samd51/variants/metro_m4/variant.cpp#L61
https://github.com/adafruit/ArduinoCore-samd/blob/samd51/variants/metro_m4/variant.h#L141

The pins chosen for `SCK`, `MOSI`, and `MISO` are `PA12`, `PA3`, and `PA15` respectively. Here's the mapping of pins to SERCOM pads from the SAMD51 datasheet:

[image]

As you know, in SPI master mode, MOSI can be on any pad (0-3), but `MISO` and `SCK` can only be on certain pads. On SAMD21, `SCK` must be on pad 1 or 3. On SAMD51, the data sheet is even more restrictive: only pad 1 available for `SCK`. (Screen shots from the datasheets are at the bottom of this issue.)

So it looks to me like we'd have to use `SERCOM4` for SPI when using these pins. I thought you couldn't use `SERCOM2`, because `SCK` on pad 0 is not possible. Is this a bug, or am I misunderstanding?

Also in the Arduino code, is there a copy/paste typo in `metro_m4/variant.h` ?
```
#define PIN_SPI_MISO         (26u)
#define PIN_SPI_MOSI         (27u)
#define PIN_SPI_SCK          (28u)
#define PERIPH_SPI           sercom2
#define PAD_SPI_TX           SPI_PAD_0_SCK_1
#define PAD_SPI_RX           SERCOM_RX_PAD_3

static const uint8_t SS      = PIN_A2 ;   
static const uint8_t MOSI = PIN_SPI_MOSI ;
static const uint8_t MISO = PIN_SPI_MISO ;
static const uint8_t SCK  = PIN_SPI_SCK ;

#define PIN_SPI1_MISO         (34u)
#define PIN_SPI1_MOSI         (35u)
#define PIN_SPI1_SCK          (36u)
#define PERIPH_SPI1           sercom4
#define PAD_SPI1_TX           SPI_PAD_0_SCK_1
#define PAD_SPI1_RX           SERCOM_RX_PAD_3

static const uint8_t SS1   = 39 ;    // HW SS isn't used. Set here only for reference.
static const uint8_t MOSI1 = PIN_SPI_MOSI ;    // <------------------did you mean PIN_SPI1_MOSI here??
static const uint8_t MISO1 = PIN_SPI_MISO ;    // <------------------etc.?
static const uint8_t SCK1  = PIN_SPI_SCK ;
```

(I discovered this because I was testing CircuitPython SPI on the Metro M4, using the standard SPI pins on the 6-pin header. There is a minor problem in the CPy code that it tries `SERCOM2` first and only realizes later that it can't use that one because of the pad restrictions. I will fix that. But that initially caused me to look at the Arduino core to see if we had made some pin assignment mistake.)

--------------------------------------------------
FYI, from the datasheets:

SAMD21 available DOPO settings:

[image]

On SAMD51, the data sheet shows only though Atmel START seems to allow the 0x1 and 0x3 values below:

[image]

ladyada, 2017-11-16:
yah it looks like SCK and MOSI need swapping on the rev B PCB?

dhalbert, 2017-11-16:
I had a chat with Scott and looked at this again. Scott sent me an email from Sept 29 where he brought up swapping SCK and MOSI.

Board schematics:
Rev A:   SCK is PA12, MOSI is PA13
Rev B&C: SCK is PA13, MOSI is PA12

So the swap did happen already on the boards.

in `variants/metro_m4/variant.cpp`, the large tabular comment still says PA12 SCK and PA13 MOSI
The `PinDescription` struct has entry 26 PA15, 27 PA12, 28 PA13, as it should be in Rev B. The comment says `// 26..28 - SPI pins (ICSP:MISO,SCK,MOSI)` but the order in the comment does _not_ correspond to the rows.

in `variant.h`, we've got:
```
#define PIN_SPI_MISO         (26u)
#define PIN_SPI_MOSI         (27u)
#define PIN_SPI_SCK          (28u)
```
which does reflect the rev B assignments (PA13 as SCK).

**So the code is right and the comments are wrong**, as far as I can tell.

Note that the `SPI1` constants may still be wrong, as noted above:

```
static const uint8_t MOSI1 = PIN_SPI_MOSI ;
static const uint8_t MISO1 = PIN_SPI_MISO ;
static const uint8_t SCK1  = PIN_SPI_SCK ;
```

And, the pin data in CircuitPython reflects rev A, not rev B. So I'll fix that.

deanm1278, 2017-11-20:
@dhalbert sorry I missed this, I think I accidentally unsubscribed from this repo. Yea the big tabular comment hasn't been updated yet. I will update soon so it matches. Sorry for the confusion

deanm1278, 2018-04-02:
old issue, misleading comments have since been deleted. Closing!

closed 2018-04-02 by deanm1278