adafruit/ArduinoCore-samd #22: feather_m0 compile error for samd51 branch

jerryneedell, 2017-11-18:
The changes made to boards.txt with 
commit 03590c5febd55be086b133a53eaedde149801df7
cause builds for the feather_m0 to fail -- see error messages below.

the sketch compiles and runs if I comment out line 44 of boards.txt
https://github.com/adafruit/ArduinoCore-samd/blob/03590c5febd55be086b133a53eaedde149801df7/boards.txt#L44

Note - the feather_m0_express variant compiles without error

Failed compile with board.txt line 44 
```Arduino: 1.8.5 (Linux), Board: "Adafruit Feather M0"

/home/jerryneedell/arduino-1.8.5/arduino-builder -dump-prefs -logger=machine -hardware /home/jerryneedell/arduino-1.8.5/hardware -hardware /home/jerryneedell/.arduino15/packages -hardware /home/jerryneedell/Arduino/hardware -tools /home/jerryneedell/arduino-1.8.5/tools-builder -tools /home/jerryneedell/arduino-1.8.5/hardware/tools/avr -tools /home/jerryneedell/.arduino15/packages -built-in-libraries /home/jerryneedell/arduino-1.8.5/libraries -libraries /home/jerryneedell/Arduino/libraries -fqbn=Adafruit:samd:adafruit_feather_m0 -vid-pid=0X239A_0X800B -ide-version=10805 -build-path /tmp/arduino_build_795374 -warnings=none -build-cache /tmp/arduino_cache_264443 -prefs=build.warn_data_percentage=75 -verbose /tmp/arduino_modified_sketch_682095/BluefruitLE_nrf51822.ino
/home/jerryneedell/arduino-1.8.5/arduino-builder -compile -logger=machine -hardware /home/jerryneedell/arduino-1.8.5/hardware -hardware /home/jerryneedell/.arduino15/packages -hardware /home/jerryneedell/Arduino/hardware -tools /home/jerryneedell/arduino-1.8.5/tools-builder -tools /home/jerryneedell/arduino-1.8.5/hardware/tools/avr -tools /home/jerryneedell/.arduino15/packages -built-in-libraries /home/jerryneedell/arduino-1.8.5/libraries -libraries /home/jerryneedell/Arduino/libraries -fqbn=Adafruit:samd:adafruit_feather_m0 -vid-pid=0X239A_0X800B -ide-version=10805 -build-path /tmp/arduino_build_795374 -warnings=none -build-cache /tmp/arduino_cache_264443 -prefs=build.warn_data_percentage=75 -verbose /tmp/arduino_modified_sketch_682095/BluefruitLE_nrf51822.ino
Using board 'adafruit_feather_m0' from platform in folder: /home/jerryneedell/Arduino/hardware/Adafruit/samd
Using core 'arduino' from platform in folder: /home/jerryneedell/Arduino/hardware/Adafruit/samd
Build options changed, rebuilding all
Detecting libraries used...
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_SAMD_FEATHER_M0 -DARDUINO_ARCH_SAMD  -DARDUINO_SAMD_ZERO -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x800B -DUSBCON '-DUSB_MANUFACTURER="Adafruit"' '-DUSB_PRODUCT="Feather M0"' "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/arduino_zero" "/tmp/arduino_build_795374/sketch/BluefruitLE_nrf51822.ino.cpp" -o "/dev/null"
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_SAMD_FEATHER_M0 -DARDUINO_ARCH_SAMD  -DARDUINO_SAMD_ZERO -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x800B -DUSBCON '-DUSB_MANUFACTURER="Adafruit"' '-DUSB_PRODUCT="Feather M0"' "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/arduino_zero" "-I/home/jerryneedell/arduino-1.8.5/libraries/Servo/src" "/tmp/arduino_build_795374/sketch/BluefruitLE_nrf51822.ino.cpp" -o "/dev/null"
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_SAMD_FEATHER_M0 -DARDUINO_ARCH_SAMD  -DARDUINO_SAMD_ZERO -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x800B -DUSBCON '-DUSB_MANUFACTURER="Adafruit"' '-DUSB_PRODUCT="Feather M0"' "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/arduino_zero" "-I/home/jerryneedell/arduino-1.8.5/libraries/Servo/src" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/libraries/Wire" "/tmp/arduino_build_795374/sketch/BluefruitLE_nrf51822.ino.cpp" -o "/dev/null"
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_SAMD_FEATHER_M0 -DARDUINO_ARCH_SAMD  -DARDUINO_SAMD_ZERO -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x800B -DUSBCON '-DUSB_MANUFACTURER="Adafruit"' '-DUSB_PRODUCT="Feather M0"' "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/arduino_zero" "-I/home/jerryneedell/arduino-1.8.5/libraries/Servo/src" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/libraries/Wire" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/libraries/SPI" "/tmp/arduino_build_795374/sketch/BluefruitLE_nrf51822.ino.cpp" -o "/dev/null"
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_SAMD_FEATHER_M0 -DARDUINO_ARCH_SAMD  -DARDUINO_SAMD_ZERO -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x800B -DUSBCON '-DUSB_MANUFACTURER="Adafruit"' '-DUSB_PRODUCT="Feather M0"' "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/arduino_zero" "-I/home/jerryneedell/arduino-1.8.5/libraries/Servo/src" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/libraries/Wire" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/libraries/SPI" "-I/home/jerryneedell/Arduino/libraries/Adafruit_BLEFirmata" "/tmp/arduino_build_795374/sketch/BluefruitLE_nrf51822.ino.cpp" -o "/dev/null"
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_SAMD_FEATHER_M0 -DARDUINO_ARCH_SAMD  -DARDUINO_SAMD_ZERO -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x800B -DUSBCON '-DUSB_MANUFACTURER="Adafruit"' '-DUSB_PRODUCT="Feather M0"' "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/arduino_zero" "-I/home/jerryneedell/ard [... cut ...]
Using library Wire at version 1.0 in folder: /home/jerryneedell/Arduino/hardware/Adafruit/samd/libraries/Wire 
Using library SPI at version 1.0 in folder: /home/jerryneedell/Arduino/hardware/Adafruit/samd/libraries/SPI 
Using library Adafruit_BLEFirmata at version 1.1.0 in folder: /home/jerryneedell/Arduino/libraries/Adafruit_BLEFirmata 
Using library Adafruit_BluefruitLE_nRF51 at version 1.9.5 in folder: /home/jerryneedell/Arduino/libraries/Adafruit_BluefruitLE_nRF51 
exit status 1
Error compiling for board Adafruit Feather M0.
```

jerryneedell, 2017-11-18:
Regarding the commit referenced above, it also made a related change to platform.txt. I did not revert that change in my testing and the compiled sketch appears to work but I am not sure what it any impact that change may have had in terms of the math libraries used since -larm_cortexM01l_math is not included but that did appear to cause any errors. 

here is the change ti platform.txt
https://github.com/adafruit/ArduinoCore-samd/commit/03590c5febd55be086b133a53eaedde149801df7#diff-d638a47d1ca0e2a5585fb34c7d7fb0f9R68

jerryneedell, 2017-11-19:
Just for completeness, I also modified platform.txt to rrevet line 68 to:
```
compiler.arm.cmsis.ldflags="-L{runtime.tools.CMSIS-4.5.0.path}/CMSIS/Lib/GCC/" -larm_cortexM0l_math
```
and the sketch compiles as well for the feather_m0 . It is probably a good thing ti include this library. This sketch may not need it, but it may be useful in other cases.

deanm1278, 2017-11-20:
hey @jerryneedell sorry about that, I've fixed by adding library paths as well to the m0 boards:
https://github.com/adafruit/ArduinoCore-samd/commit/5d4285e2d606ea3022bb5fe1bd913ede919809fa

closed 2017-11-20 by deanm1278

jerryneedell, 2017-11-20:
Thanks - works fine now!