adafruit/ArduinoCore-samd #23: Serial Buffer Size

fabltd, 2017-12-19:
Where is define SERIAL_BUFFER_SIZE located. 

Can you confirm:

"Define constants and variables for buffering incoming serial data.  We're
 using a ring buffer (I think)" 

Are we using ring buffer or not?

deanm1278, 2017-12-19:
yes we are using a ring buffer for incoming serial data.

`#define SERIAL_BUFFER_SIZE 164`
is located in RingBuffer.h

https://github.com/adafruit/ArduinoCore-samd/blob/master/cores/arduino/RingBuffer.h

closed 2017-12-19 by deanm1278

fabltd, 2017-12-19:
Suggest the comment is changed from (I think) !

So buffer is 128 k. Can this be increased to 256k

deanm1278, 2017-12-19:
The buffer is 164 bytes. The samd21 doesn't have enough RAM for a 256k byte buffer.
Note that this is for UART serial data. For serial data via USB a different buffer is used.

fabltd, 2017-12-19:
Thanks.

Can it be increased from 128? Or is this the max. Its slow on a 30KB
transfer.

-- 

Jeremy Turner

Google Cloud Platform Authorised Trainer

Director @fabcontrols

FAB Controls Ltd

Unit 6,St Albans Enterprise Centre

Long Spring, Porters Wood

St Albans, Hertfordshire

AL3 6EN

*Phone:* +44(0) 1727 238 894

Cloud implmentation & ‘Internet of Things’ (IoT) solutions provider