adafruit/ArduinoCore-samd #25: unable to build M4 sketches

jerryneedell, 2017-12-23:
I had followed the steps in the samd51 branch and have been able to compile sketches for my Metro M4 for several months. I tried today and got a bunch of error at the compile stage trying to compile blink.ino
I updated from the samd51 branch repo and re-installed the CMSIS Device directory. The compile errors all went away but now I get linker error. see attached log file.
I get the same result on MacOS and Linux.

Any idea what to d next? Thanks
```Arduino: 1.8.5 (Linux), Board: "Adafruit Metro M4 (SAMD51)"

/home/jerryneedell/arduino-1.8.5/arduino-builder -dump-prefs -logger=machine -hardware /home/jerryneedell/arduino-1.8.5/hardware -hardware /home/jerryneedell/.arduino15/packages -hardware /home/jerryneedell/Arduino/hardware -tools /home/jerryneedell/arduino-1.8.5/tools-builder -tools /home/jerryneedell/arduino-1.8.5/hardware/tools/avr -tools /home/jerryneedell/.arduino15/packages -built-in-libraries /home/jerryneedell/arduino-1.8.5/libraries -libraries /home/jerryneedell/Arduino/libraries -fqbn=Adafruit:samd:adafruit_metro_m4 -ide-version=10805 -build-path /tmp/arduino_build_873806 -warnings=none -build-cache /tmp/arduino_cache_612797 -prefs=build.warn_data_percentage=75 -verbose /home/jerryneedell/arduino-1.8.5/examples/01.Basics/Blink/Blink.ino
/home/jerryneedell/arduino-1.8.5/arduino-builder -compile -logger=machine -hardware /home/jerryneedell/arduino-1.8.5/hardware -hardware /home/jerryneedell/.arduino15/packages -hardware /home/jerryneedell/Arduino/hardware -tools /home/jerryneedell/arduino-1.8.5/tools-builder -tools /home/jerryneedell/arduino-1.8.5/hardware/tools/avr -tools /home/jerryneedell/.arduino15/packages -built-in-libraries /home/jerryneedell/arduino-1.8.5/libraries -libraries /home/jerryneedell/Arduino/libraries -fqbn=Adafruit:samd:adafruit_metro_m4 -ide-version=10805 -build-path /tmp/arduino_build_873806 -warnings=none -build-cache /tmp/arduino_cache_612797 -prefs=build.warn_data_percentage=75 -verbose /home/jerryneedell/arduino-1.8.5/examples/01.Basics/Blink/Blink.ino
Using board 'adafruit_metro_m4' from platform in folder: /home/jerryneedell/Arduino/hardware/Adafruit/samd
Using core 'arduino' from platform in folder: /home/jerryneedell/Arduino/hardware/Adafruit/samd
Detecting libraries used...
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g3 -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_METRO_M4 -DARDUINO_ARCH_SAMD  -D__SAMD51J19A__ -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8020 -DUSBCON '-DUSB_MANUFACTURER="Adafruit LLC"' '-DUSB_PRODUCT="Adafruit Metro M4"' -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/metro_m4" "/tmp/arduino_build_873806/sketch/Blink.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g3 -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_METRO_M4 -DARDUINO_ARCH_SAMD  -D__SAMD51J19A__ -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8020 -DUSBCON '-DUSB_MANUFACTURER="Adafruit LLC"' '-DUSB_PRODUCT="Adafruit Metro M4"' -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/metro_m4" "/tmp/arduino_build_873806/sketch/Blink.ino.cpp" -o "/tmp/arduino_build_873806/preproc/ctags_target_for_gcc_minus_e.cpp"
"/home/jerryneedell/arduino-1.8.5/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino_build_873806/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g3 -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_METRO_M4 -DARDUINO_ARCH_SAMD  -D__SAMD51J19A__ -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8020 -DUSBCON '-DUSB_MANUFACTURER="Adafruit LLC"' '-DUSB_PRODUCT="Adafruit Metro M4"' -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/metro_m4" "/tmp/arduino_build_873806/sketch/Blink.ino.cpp" -o "/tmp/arduino_build_873806/sketch/Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g3 -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_METRO_M4 -DARDUINO_ARCH_SAMD  -D__SAMD51J19A__ -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8020 -DUSBCON '-DUSB_MANUFACTURER="Adafruit LLC"' '-DUSB_PRODUCT="Adafruit Metro M4"' -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/metro_m4" "/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/metro_m4/variant.cpp" -o "/tmp/arduino_build_873806/core/variant.cpp.o"
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc" -c -g3 -x assembler-with-cpp -MMD -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_METRO_M4 -DARDUINO_ARCH_SAMD  -D__SAMD51J19A__ -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8020 -DUSBCON '-DUSB_MANUFACTURER="Adafruit LLC"' '-DUSB_PRODUCT="Adafruit Metro M4"' -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/metro_m4" "/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino/pulse_asm.S" -o "/tmp/arduino_build_873806/core/pulse_asm.S.o"
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc" -mcpu=cortex-m4 -mthumb -c -g3 -Os -w -std=gnu11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -MMD -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_METRO_M4 -DARDUINO_ARCH_SAMD  -D__SAMD51J19A__ -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8020 -DUSBCON '-DUSB_MANUFACTURER="Adafruit LLC"' '-DUSB_PRODUCT="Adafruit Metro M4"' -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/metro_m4" "/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino/WInterrupts.c" -o "/tmp/arduino_build_873806/core/WInterrupts.c.o"
[... cut ...]
/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: error: /tmp/arduino_build_873806/core/core.a(PluggableUSB.cpp.o) uses VFP register arguments, /tmp/arduino_build_873806/Blink.ino.elf does not
/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /tmp/arduino_build_873806/core/core.a(PluggableUSB.cpp.o)
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Adafruit Metro M4 (SAMD51).```

jerryneedell, 2018-01-04:
FYI - I just did a complete re-install of this and I'm still getting the same errors. Any suggestions. Can yo compile blink,ino for a Metro M4 Express?

deanm1278, 2018-01-04:
Hey @jerryneedell sorry about that. Try again with most recent samd51 branch?

jerryneedell, 2018-01-04:
wohoo! fixed!

closed 2018-01-04 by jerryneedell