adafruit/ArduinoCore-samd #26: arm_calc_2pow() in core/math_helper.c is silly

WestfW, 2018-02-01:
it does a loop of multiplies by 2 when AFAICT, a shift should work  much better.

I've submitted a similar complaint against the  ARM/CMSIS repository, but there math_helper.c is only used in examples and test code, so they may not be very motivated to change it.  (Then again, if you're actually USING these functions on a CM0, you're likely to be pretty unhappy, too.)

ladyada, 2018-02-01:
hiya this is a fork of arduino's core, we recommend opening an issue there, as we do keep up with upstream!

dhalbert, 2018-02-01:
I see the function in ARM/CMSIS, bu there is no `math_helper.c` file in this repo or in upstream, or an implementation of that function.

deanm1278, 2018-02-01:
hey @WestfW have you tested the asm output the ARM compiler generates for `x*2` vs `x<<1` on an unsigned integer data type? I'd imagine it might catch something like that?

@dhalbert it's in the SAMD51 branch, added when I was adding support for the teensy audio library. It may get removed before that branch is merged w/ mainline

WestfW, 2018-02-01:
1) oops.   I didn't realize that different branches wound up with the same issue list.  This is only in the SAMD51 branch, AFAIK.

2) I'm sure that the compiler with optimize x*2 vs x<<1, but in this case the code has a loop with a variable counter, so I'm pretty sure no optimization is possible:

```
 uint32_t i, val = 1;
  for (i = 0; i < numShifts; i++) {
      val = val * 2;
    }	
return(val);
```

deanm1278, 2018-02-01:
I was curious so I just gave it a try:

```
uint32_t fakeFn(uint8_t numShifts){
    uint32_t i, val = 1;
     for (i = 0; i < numShifts; i++) {
         val = val * 2;
       }
   return(val);
}

uint32_t fakeFn2(uint8_t numShifts){
    uint32_t i, val = 1;
     for (i = 0; i < numShifts; i++) {
         val = val << 1;
       }
   return(val);
}

// the loop routine runs over and over again forever:
void loop() {
    for(int i=0; i<12; i++){
        uint32_t g = fakeFn(i);
        Serial.println(g);
        g = fakeFn2(i);
        Serial.println(g);
    }
}
```

ASM output looks identical:
```
_Z6fakeFnh:
.LFB228:
	.loc 1 65 0
	.cfi_startproc
	@ args = 0, pretend = 0, frame = 16
	@ frame_needed = 1, uses_anonymous_args = 0
	@ link register save eliminated.
	push	{r7}
	.cfi_def_cfa_offset 4
	.cfi_offset 7, -4
	sub	sp, sp, #20
	.cfi_def_cfa_offset 24
	add	r7, sp, #0
	.cfi_def_cfa_register 7
	mov	r3, r0
	strb	r3, [r7, #7]
.LBB2:
	.loc 1 66 0
	movs	r3, #1
	str	r3, [r7, #8]
	.loc 1 67 0
	movs	r3, #0
	str	r3, [r7, #12]
	b	.L5
.L6:
	.loc 1 68 0 discriminator 2
	ldr	r3, [r7, #8]
	lsls	r3, r3, #1
	str	r3, [r7, #8]
	.loc 1 67 0 discriminator 2
	ldr	r3, [r7, #12]
	adds	r3, r3, #1
	str	r3, [r7, #12]
.L5:
	.loc 1 67 0 is_stmt 0 discriminator 1
	ldrb	r2, [r7, #7]	@ zero_extendqisi2
	ldr	r3, [r7, #12]
	cmp	r2, r3
	bhi	.L6
	.loc 1 70 0 is_stmt 1
	ldr	r3, [r7, #8]
.LBE2:
	.loc 1 71 0
	mov	r0, r3
	adds	r7, r7, #20
	mov	sp, r7
	@ sp needed
	ldr	r7, [sp], #4
	bx	lr
	.cfi_endproc
```

vs: 
```
_Z7fakeFn2h:
.LFB229:
	.loc 1 73 0
	.cfi_startproc
	@ args = 0, pretend = 0, frame = 16
	@ frame_needed = 1, uses_anonymous_args = 0
	@ link register save eliminated.
	push	{r7}
	.cfi_def_cfa_offset 4
	.cfi_offset 7, -4
	sub	sp, sp, #20
	.cfi_def_cfa_offset 24
	add	r7, sp, #0
	.cfi_def_cfa_register 7
	mov	r3, r0
	strb	r3, [r7, #7]
.LBB3:
	.loc 1 74 0
	movs	r3, #1
	str	r3, [r7, #8]
	.loc 1 75 0
	movs	r3, #0
	str	r3, [r7, #12]
	b	.L9
.L10:
	.loc 1 76 0 discriminator 2
	ldr	r3, [r7, #8]
	lsls	r3, r3, #1
	str	r3, [r7, #8]
	.loc 1 75 0 discriminator 2
	ldr	r3, [r7, #12]
	adds	r3, r3, #1
	str	r3, [r7, #12]
.L9:
	.loc 1 75 0 is_stmt 0 discriminator 1
	ldrb	r2, [r7, #7]	@ zero_extendqisi2
	ldr	r3, [r7, #12]
	cmp	r2, r3
	bhi	.L10
	.loc 1 78 0 is_stmt 1
	ldr	r3, [r7, #8]
.LBE3:
	.loc 1 79 0
	mov	r0, r3
	adds	r7, r7, #20
	mov	sp, r7
	@ sp needed
	ldr	r7, [sp], #4
	bx	lr
	.cfi_endproc
```

WestfW, 2018-02-02:
I wanted:

`static inline uint32_t fastPow2(uint8_t numShifts) { return ((uint32_t)1)<<numShifts; }`

closed 2018-04-02 by deanm1278