adafruit/ArduinoCore-samd #33: Bootloader for ATSAME54N20

freemovers, 2018-04-04:
I'm working on a custom board where I like to Arduino to create the programs. I customized BOSSAC already to accept the SAME54, and I am able to upload sketches through Arduino using the METRO_M4_sam_ba.hex for the bootloader. I like to compile a custom bootloader for the ATSAME54N20, but have difficulties compiling the source with MSYS. I always get the following error, even after adding the board ID:
Compiling bootloader using
BASE PATH = C:\Users\svandebor\AppData\Local/Arduino15/packages/arduino
GCC PATH = C:\Users\svandebor\AppData\Local/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-
board id = METRO_M4
----------------------------------------------------------
Compiling board_driver_led.c to build/board_driver_led.o
"C:\Users\svandebor\AppData\Local/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc" -mthumb -mcpu=cortex-m4 -c -std=gnu99 -MD -MP -MF "build/board_driver_led.d" -MT"build/board_driver_led.d" -MT"build/board_driver_led.o" -ffunction-sections -fdata-sections -nostdlib -nostartfiles --param max-inline-insns-single=500 -Os -DDEBUG=0 -D -DBOARD_ID_METRO_M4 -I"C:\Users\svandebor\AppData\Local/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" -I"C:\Users\svandebor\AppData\Local/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" board_driver_led.c -o build/board_driver_led.o
<command-line>:0:1: error: macro names must be identifiers
In file included from board_driver_led.h:24:0,
                 from board_driver_led.c:20:
board_definitions.h:25:4: error: #error You must define a BOARD_ID and add the corresponding definitions in board_definitions.h
   #error You must define a BOARD_ID and add the corresponding definitions in board_definitions.h
    ^
board_driver_led.c:22:1: error: unknown type name 'uint8_t'
 volatile uint8_t ledKeepValue = 0;
 ^
board_driver_led.c:23:1: error: unknown type name 'uint8_t'
 volatile uint8_t ledTargetValue = 20;
 ^
board_driver_led.c:24:1: error: unknown type name 'int8_t'
 volatile int8_t ledDirection = 1;
 ^
make: *** [build/board_driver_led.o] Error 1

I have no issue compiling the "zero" bootloaders.

freemovers, 2018-04-04:
I added back the board ID and board chip in the **makefile**, and it seems to work now. 

_# Boards definitions
BOARD_ID?=METRO_M4
BOARD_CHIP?=__SAMD51J19A___

Are you usually adding these through the command line?

freemovers, 2018-04-04:
Thank your for the quick response. I think I got it, I was even able to compile for the ATSAME54N20 after adding some lines to the CMSIS files. I will look into the UF2 bootloaders as well, haven't used those yet, but looks promising to make it even more user friendly.

deanm1278, 2018-04-04:
here's our uf2 repo: https://github.com/adafruit/uf2-samd21

closed 2018-04-04 by deanm1278

ladyada, 2018-04-04:
i second recommend UF2, its bossa compatible but has drag n drop and webUSB support (in the future), its what we're using going forward for SAMD chips

freemovers, 2018-04-04:
I’m looking into the UF2 right now, try to find out what will be involved to get it going for the ATSAME54N20. It looks pretty simple compared to the SAMBA bootloader, but I might miss something here J. I will just download, make some changes, and compile.