adafruit/ArduinoCore-samd #37: Cache?

WestfW, 2018-04-27:
I don't see any code in the core, or CMSIS, to initialize the Cache Memory on the SAMD51
(CMCC)

WestfW, 2018-04-27:
>  if you have an example code...
Sorry; no actual experience.  I read the forum message about "slower than expected" and was thinking "well, the flash memory isn't any faster, but ... isn't there supposed to be cache or an accelerator or something - I wonder how it's configured?"  And then I didn't find anything in the source, but confirmed that it's documented in the datasheet.  Sort-of.
I'm particularly bothered because it's not clear whether the cache memory is something that needs to be taken out of the normal RAM space (ie in libker map) if you're using it.  I guess if cache is disabled, the memory becomes the "Tightly Coupled Memory" mentioned (but not described very well) elsewhere in the datasheet.

dhalbert, 2018-04-28:
It does not appear to me that cache is taken from normal RAM space. The Tightly Coupled Memory functionality is not really explained at all.

I tried some experiments in which I changed CMCC.CFG.CSIZESW, which lets you set the cache size from 0 to 64KB. However, doing so, did not change the size of the cache reported in CMCC.TYPE.CSIZE. So maybe this is latent functionality for chips with potentially larger caches.

WestfW, 2018-04-28:
Yeah; my interpretation is that the CM4 CMCC can have up to 64k of cache, but the SAMD51 only has 4k, and you can configure it to use less (in which case TCM appears ... somewhere.)

I was wondering whether it might make sense to disable the Data Cache (since the internal RAM is single-cycle anyway), potentially freeing more cache for instructions.

I'd expect the cache to be important when you start using the QSPI flash, especially if you're going to put XIP code in there.