adafruit/ArduinoCore-samd #43: SPI is not working properly

aryelevin, 2018-04-30:
I'm using riverdi Arduino shield which work perfect with the same code on Arduino zero, but in the metro m4 I get a blank white screen and the initialisation of the screen is not working.

I've tried to lower the speed, with no luck.
The same code works perfect on the Arduino zero without any modification.

aryelevin, 2018-05-01:
OK, after further investigation, I recognised some difference between the SAMD51 to the SAMD21 MCUs SPI performance.
Finally I've got my Riverdi LCD working by adding a resistor on each of the SPI lines and the CS line (as close as possible to the FT813Q controller).
I assume this is duo to some differences on the pins I/O characteristics on the SAMD51.

I don't know if it should be closed or further investigated.

Thanks for the library.

deanm1278, 2018-05-01:
hi @aryelevin is this a series termination resistor you're talking about or a pullup resistor?
We may be increasing the drive strength on the SPI bus and/or all IO pins if there are issues with devices.

aryelevin, 2018-05-02:
Hi @deanm1278 ,
Its a series termination resistors, not pull-ups.

Let me know if you need my help with testing your changes.

aryelevin, 2018-05-16:
Hi @deanm1278 .
Any progress with it?
I've designed my own board with WINC1500 and SPI SD card, and the Riverdi LCD connector, and the issue is that the display is not working properly, and the WINC1500 is also not working properly, only the SD card it working.

Can you tell me how to increase the drive strength on the SPI bus and/or all IO pins if there are issues with devices? So I will be able to see if its works?

Thank you very much in advance.

Arye

ladyada, 2018-05-16:
hiya 
we tested the metro m4 with winc1500 shield and it worked fine for us :)
this isnt something we're working on right now, if you find a solution please submit a PR but we have no ETA for when we'd look at it or fix it!

dhalbert, 2018-05-16:
To set the drive strength to 7ma, you set the `PINCFGxx.DRVSTR` bit. This is in the same register for each pin as `PULLEN` and `INEN`, so if you search for those symbols, you can how to choose the register to use.

aryelevin, 2018-05-16:
Thanks @dhalbert ,

I will dig into this now and will report.

ladyada, 2019-04-22:
closing for lack of activity - we've been using SPI a lot without issue (?)

closed 2019-04-22 by ladyada

aryelevin, 2019-04-22:
Will try to disable it I'm my code, and if there's issues, will reopen.
Thanks