adafruit/ArduinoCore-samd #47: Bugs in the SAMD51 code

Misfittech, 2018-05-27:
In the void SystemInit( void )  the code sets the Flash wait states to 0, however at 120Mhz it needs to be 5. 

In the Reset_Handler() the if statement before clearing the zero section is incorrect, cut and paste error from the code above.   It should be:
if ((&__bss_start__ != &__bss_end__))

dhalbert, 2018-05-27:
SAMD51 can figure out the number of wait states automatically: see NVMCTRL.CTRLA.AUTOWS. This bit is set automatically on reset, so you don't normally need to specify the number of wait states. If this bit is set, NVMCTRL.CTRLA.RWS is ignored.

Are we looking at the same code for `Reset_Handler()`? I'm looking in `bootloaders/feather/board_startup.c`. The current code is:
```
 void Reset_Handler( void )
{
  uint32_t *pSrc, *pDest;

  /* Initialize the initialized data section */
  pSrc = &__etext;
  pDest = &__data_start__;

  if ( (&__data_start__ != &__data_end__) && (pSrc != pDest) )
  {
    for (; pDest < &__data_end__ ; pDest++, pSrc++ )
    {
      *pDest = *pSrc ;
    }
  }

  /* Clear the zero section */
  if ( &__bss_start__ != &__bss_end__ )
  {
    for ( pDest = &__bss_start__ ; pDest < &__bss_end__ ; pDest++ )
    {
      *pDest = 0ul ;
    }
  }

//  board_init(); // will be done in main() after app check

  /* Initialize the C library */
//  __libc_init_array();

  main();

  while (1);
}
```
I believe you are saying the seond `if` should be
```
  if ((&__bss_start__ != &__bss_end__))
```
but are you just saying it should an extra set of parentheses? The code looks OK to me. Maybe @deanm1278 has something to add.

deanm1278, 2018-05-28:
yup the processor automatically determines the number of wait states so the set number of wait states is not used.
Can you point to the file and line number of the zero section issue and describe the bug that it is causing?

Misfittech, 2018-05-28:
in cortex_handlers.c

/* This is called on processor reset to initialize the device and call
main() */
void Reset_Handler(void)
{
  uint32_t *pSrc, *pDest;

  /* Initialize the initialized data section */
  pSrc = &__etext;
  pDest = &__data_start__;

  if ((&__data_start__ != &__data_end__) && (pSrc != pDest)) {
    for (; pDest < &__data_end__; pDest++, pSrc++)
      *pDest = *pSrc;
  }

  /* Clear the zero section */
*  if ((&__data_start__ != &__data_end__) && (pSrc != pDest)) {*
    for (pDest = &__bss_start__; pDest < &__bss_end__; pDest++)
      *pDest = 0;
  }

#if defined(__FPU_USED) && defined(__SAMD51__)
/* Enable FPU */
SCB->CPACR |= (0xFu << 20);
__DSB();
__ISB();
#endif

  SystemInit();

  main();

  while (1)
    ;
}

This line appears to be in error:
  /* Clear the zero section */
*  if ((&__data_start__ != &__data_end__) && (pSrc != pDest)) {*

I have not seen a problem in the wild, only by reviewing code did I see
it.  Most likely it will never cause a problem and may have been there for
awhile.

Trampas

deanm1278, 2018-05-28:
ah ok I see what you mean. Without some digging I can't be entirely sure if the person who wrote this wrote it that way intentionally or not. Since this is forked from the arduino/ArduinoCore-samd repo (https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/cortex_handlers.c) I would suggest you make an issue or PR there, and this repo will get any changes when we merge upstream :)

Closing for now since it doesn't seem to be causing any immediate problems.

closed 2018-05-28 by deanm1278