adafruit/ArduinoCore-samd #48: I2S does not compile on SAMD51 variants

blanu, 2018-06-14:
When the board type is set to Adafruit Metro M4 (SAMD51), including <I2S.h> causes the following compiler error to occur:

.../Library/Arduino15/packages/adafruit/hardware/samd/1.2.0/libraries/I2S/src/I2S.cpp:552:47: error: 'PIN_I2S_SD' was not declared in this scope
 I2SClass I2S(I2S_DEVICE, I2S_CLOCK_GENERATOR, PIN_I2S_SD, PIN_I2S_SCK, PIN_I2S_FS);

This value appears to be defined in the variants/ files. It is defined for the feather_m0, feather_m0, metro_m0_express, itsybitsy_m0, arduino_zero, mkrfox1200, and mkr1000 variants. 

The feather_m4 and metro_m4 variants define PIN_I2S_SDI and PIN_I2S_SDO, but not PIN_I2S_SD.

ladyada, 2018-06-14:
yeah we dont support the default I2S library and we also had some issues with it on M0

do you mind trying
https://github.com/adafruit/Adafruit_ZeroI2S/tree/no-asf-rewrite
you'll need to download/install manually but that branch should be M4 compatible and happy!

ladyada, 2018-06-14:
actually, that should have been merged so...try Master. but yeah, you'll probably be happier using that OR you can use our Audio library adaptation for M4!
https://github.com/adafruit/Audio

closed 2018-06-14 by ladyada