adafruit/ArduinoCore-samd #51: Speed up ADC...

WestfW, 2018-06-24:
The Adafruit SAMD51 boards can be seen as an opportunity to speed up the Analog to Digital Converter a great deal, without running into compatibility problems with the Arduino SAMD21 core (which has some "issues" of its own.  See:  https://github.com/arduino/ArduinoCore-samd/issues/327
http://www.avdweb.nl/arduino/libraries/fast-10-bit-adc.html ) or "existing code expectations.)

Current code reads:

```
    GCLK->PCHCTRL[ADC0_GCLK_ID].reg = GCLK_PCHCTRL_GEN_GCLK1_Val | (1 << GCLK_PCHCTRL_CHEN_Pos); //use clock generator 1 (48Mhz)
	
	ADC0->CTRLA.bit.PRESCALER = ADC_CTRLA_PRESCALER_DIV256_Val;
    :
ADC0->SAMPCTRL.reg = 0x3f; // Set max Sampling Time Length
```

Which means that the ADC clock will run at 48M/256 = 187500Hz, for an effective conversion time of about 60us.  In addition, the long sample time (64 clocks! see the above issue, and note that either the D51 is missing the factor or 1/2 that the D21 has, or one of the datasheets is wrong) further lengthens the overall ADC "propagation delay" to ~400us, which is ... really slow for an ADC peripheral that Atmel advertises as being capable of 1M samples/s.

I'd suggest that the prescaler be changed to 2 (minimum value), giving a 24MHz ADC clock (1Msps is achieved with a 32MHz ADC clock, at max resolution), and the SAMPCTRL register be set to ... a small integer, like 4. (There is discussion of reasonable sample times in the SAMD21 issue, and near Figure 54-5 ("54.10.4 Analog-to-Digital Converter (ADC) Characteristics" Section.)

In addition, you could leave the ADC enabled with the Reference Voltage set appropriately, which would further cut conversion time in half by no longer needing to throw away the "first ADC read after the reference has been changed.)

deanm1278, 2018-10-01:
ok updated! https://github.com/adafruit/ArduinoCore-samd/commit/00dd128b04462d5bd3f235fea8252a1f9bd6ba1e

it is now set to 1/ ( 1/(120MHz/16) ) * (30 + 1) = 241,935 Samples/Sec

closed 2018-10-01 by deanm1278

deanm1278, 2018-10-01:
slight change of plans I've set the adc sampling freq on both boards to 250k SPS
https://github.com/adafruit/ArduinoCore-samd/commit/d2ef3198333bca649ef1a9a68914e3a9abace2e6