adafruit/ArduinoCore-samd #55: Missing VARIANT_GCLK1_FREQ in variant.h - ArduinoZeroI2S fails to compile

blanu, 2018-07-24:
The following lines defined in the metro_m4 and itsybitsy_m4 variants are missing from the feather_m4 variant.h. VARIANT_GCLK1_FREQ is required by the Arduino_ZeroI2S library.

#define VARIANT_GCLK0_FREQ (120000000UL)
#define VARIANT_GCLK1_FREQ (48000000UL)
#define VARIANT_GCLK2_FREQ (100000000UL)

deanm1278, 2018-09-10:
fixed! https://github.com/adafruit/ArduinoCore-samd/commit/a8e1302151688d0d8a728ad469a45c6acd1f2077

closed 2018-09-10 by deanm1278

AzureEmpyrean, 2018-10-31:
I just downloaded the audio library and zero 
I2s library and was having this same issue for some reason. Adding in those lines seem to have fixed the compiling issue! I'll just have to test it once I get home