adafruit/ArduinoCore-samd #56: math_helper.h unknown file type uint32

SimGas, 2018-07-27:
Hi,

I wanted to debug the feather M0 using Atmel Studio. Building the basic arduino blink sketch caused the build to fail with approx 643 errors: "unknown file type uint32_t" in math_helper. Can you please fix the math_helper file so that it includes the file type definition on its own?

Thanks
Simon

deanm1278, 2018-07-27:
Hi @SimGas please set up Atmel Studio according to this guide:
https://learn.adafruit.com/proper-step-debugging-atsamd21-arduino-zero-m0?view=all#paths-and-optimizations

SimGas, 2018-07-27:
Hi,

 

I did exactly as described in this guide. But the error remains until deleting of math_helper.h

 

https://learn.adafruit.com/proper-step-debugging-atsamd21-arduino-zero-m0?view=all#paths-and-optimizations