adafruit/ArduinoCore-samd #58: Metro M4 SPI / SdFat performance issue

SapientHetero, 2018-08-08:
I just started working with the Metro M4. I mounted an Adafruit WINC1500 shield and ported Arduino-based software I wrote for a Mega to it expecting huge performance improvements, only to find that it crawls. I traced the slowdown to portions of my code that communicate with the shield, so I ran the SdFat bench example program to quantitatively compare the M4's performance to that of the Mega using the same shield, same software and same freshly formatted micro-SD card for both tests. The M4 performance on this test was on the order of 28x slower for writes and 50x slower for reads.

I haven't yet done quantitative benchmarking on WiFi performance but the software I ported to the M4 reports on its maximum loop time. When not actively serving pages the max loop time on the Mega is 2ms; on the M4, it's nearly 900ms. I've isolated the slowdown to a function that does WiFi communication, which leads me to believe the issue is SPI-related.

I've attached the version of SdFat bench I used. I modified it to use "SPI_FULL_SPEED" after my first run using the original SPI_HALF_SPEED setting produced poor results, but the results were virtually identical at each setting.

Any insights you might have into this would be deeply appreciated.

bench.zip

MEGA 2560 performance
Use a freshly formatted SD for best performance.

Type any character to start
FreeStack: 6711
Type is FAT32
Card size: 31.91 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: SC32G
Version: 8.0
Serial number: 0X111A0968
Manufacturing date: 2/2012

File size 5 MB
Buffer size 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
281.23,27624,1572,1813
284.88,95152,1540,1790

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
507.68,3108,988,1001
507.84,1992,988,1001

Metro M4 performance
Use a freshly formatted SD for best performance.

Type any character to start
FreeStack: 192112
Type is FAT32
Card size: 31.91 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: SC32G
Version: 8.0
Serial number: 0X111A0968
Manufacturing date: 2/2012

File size 5 MB
Buffer size 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
10.30,344156,49334,49677

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
10.69,95857,47883,47897
10.69,95857,47883,47897

Done
[bench.zip](https://github.com/adafruit/ArduinoCore-samd/files/2271529/bench.zip)

deanm1278, 2018-08-09:
Hey do you have any SPI traces you could post of the Mega vs the Metro M4 so we could see what's going on?

SapientHetero, 2018-08-09:
Sadly, I have no oscilloscope or friends who own one. I tried running this test with the same shield, same SanDisk Ultra 32GB SDHC UHS-I micro-SD card and an Arduino DUE and got the following result: 

Use a freshly formatted SD for best performance.

Type any character to start
FreeStack: 91576
Can't access SD card. Do not reformat.
SD errorCode: 0X50,0X6

According to SdInfo.h, 0x50 is a SD_CARD_ERROR_READ; not sure how to interpret the 0x6 yet.

I also ran SdFormatter on the DUE and it tells me "error: cardSize".

After getting this error I moved the shield back to a Mega and it works fine.

I buzzed out the SPI connections on my shield with a meter and none seem to have excessive resistance. Both the DUE and the Metro M4 are 3.3V devices, while the Mega runs at 5V. Could it have something to do with this?

SapientHetero, 2018-08-10:
I have a breadboard setup I'd been using for testing in which I use jumper wires. With the Mega I get very nearly the same performance in this environment that I get with the WINC1500 shield.

I wired the breakout-board SD card reader from that setup to my Metro M4 so it is the only SPI device attached to see if the performance problem was a bus-sharing problem. Ran SdFat 'bench' sketch again and got results comparable to those seen with WINC1500 (see below). 

I gather from reading other posts that the Metro M4 was thoroughly tested with up to 3 SPI devices at a time, so I'm beginning to wonder if my particular Metro is defective.

*** Metro M4 SdFat bench results with breadboard SD card reader ***
Use a freshly formatted SD for best performance.

Type any character to start
FreeStack: 127064
Type is FAT32
Card size: 31.91 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: SC32G
Version: 8.0
Serial number: 0X111A0968
Manufacturing date: 2/2012

FILE_SIZE_MB = 5
BUF_SIZE = 65536 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
10.94,6039334,6012943,6013321
10.94,6039334,6012943,6013320

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
10.98,5989731,5989727,5989798
10.98,5989731,5989727,5989756

Done

ryan-feeley, 2018-08-11:
I've been using a version of the SPI library that I modified with the metro M4 board + SdFat, and it works very well. 

Unfortunately Adafruit hasn't kept this fork up-to-date with the upstream (perhaps they have a good reason), so I don't know that it would make any sense to even try a pull request. Here is my SPI.h, if you're interested it trying it. I can't promise that you'll just be able to drop this in, but it might point you in the right direction.

SPI.h

```
/*
 * SPI Master library for Arduino Zero.
 * Copyright (c) 2015 Arduino LLC
 *
 * This library is free software; you can redistribute it and/or
 * modify it under the terms of the GNU Lesser General Public
 * License as published by the Free Software Foundation; either
 * version 2.1 of the License, or (at your option) any later version.
 *
 * This library is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 * Lesser General Public License for more details.
 *
 * You should have received a copy of the GNU Lesser General Public
 * License along with this library; if not, write to the Free Software
 * Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
 */

#ifndef _SPI_H_INCLUDED
#define _SPI_H_INCLUDED

#include <Arduino.h>

// SPI_HAS_TRANSACTION means SPI has
//   - beginTransaction()
//   - endTransaction()
//   - usingInterrupt()
//   - SPISetting(clock, bitOrder, dataMode)
#define SPI_HAS_TRANSACTION 1

// SPI_HAS_NOTUSINGINTERRUPT means that SPI has notUsingInterrupt() method
#define SPI_HAS_NOTUSINGINTERRUPT 1

#define SPI_MODE0 0x02
#define SPI_MODE1 0x00
#define SPI_MODE2 0x03
#define SPI_MODE3 0x01

#if defined(ARDUINO_ARCH_SAMD)
# if defined(__SAMD51__)
  // The datasheet specifies a typical SPI SCK period (tSCK) of 58 ns,
  // see note 10 following "Table 54-52. SPI Timing Characteristics and Requirements",
  // which translates into a maximum SPI clock of 17.2 MHz.
  // Conservatively, the divider is set for a 12 MHz maximum SPI clock.
  #define SPI_MIN_CLOCK_DIVIDER (uint8_t)(1 + ((F_CPU - 1) / 12000000))
# else
  // The datasheet specifies a typical SPI SCK period (tSCK) of 42 ns,
  // see "Table 36-48. SPI Timing Characteristics and Requirements",
  // which translates into a maximum SPI clock of 23.8 MHz.
  // Conservatively, the divider is set for a 12 MHz maximum SPI clock.
  #define SPI_MIN_CLOCK_DIVIDER (uint8_t)(1 + ((F_CPU - 1) / 12000000))
# endif
#endif

class SPISettings {
  public:
  SPISettings(uint32_t clock, BitOrder bitOrder, uint8_t dataMode) {
    if (__builtin_constant_p(clock)) {
      init_AlwaysInline(clock, bitOrder, dataMode);
    } else {
      init_MightInline(clock, bitOrder, dataMode);
    }
[... 377 lines cut ...]
   *   - SercomRXPad
   */
  #ifndef PERIPH_SPI
    #define PERIPH_SPI           sercom4
    #define PAD_SPI_TX           SPI_PAD_2_SCK_3
    #define PAD_SPI_RX           SERCOM_RX_PAD_0
  #endif // PERIPH_SPI
  SPIClass SPI (&PERIPH_SPI,  PIN_SPI_MISO,  PIN_SPI_SCK,  PIN_SPI_MOSI,  PAD_SPI_TX,  PAD_SPI_RX);
#endif
#if SPI_INTERFACES_COUNT > 1
  SPIClass SPI1(&PERIPH_SPI1, PIN_SPI1_MISO, PIN_SPI1_SCK, PIN_SPI1_MOSI, PAD_SPI1_TX, PAD_SPI1_RX);
#endif
#if SPI_INTERFACES_COUNT > 2
  SPIClass SPI2(&PERIPH_SPI2, PIN_SPI2_MISO, PIN_SPI2_SCK, PIN_SPI2_MOSI, PAD_SPI2_TX, PAD_SPI2_RX);
#endif
#if SPI_INTERFACES_COUNT > 3
  SPIClass SPI3(&PERIPH_SPI3, PIN_SPI3_MISO, PIN_SPI3_SCK, PIN_SPI3_MOSI, PAD_SPI3_TX, PAD_SPI3_RX);
#endif
#if SPI_INTERFACES_COUNT > 4
  SPIClass SPI4(&PERIPH_SPI4, PIN_SPI4_MISO, PIN_SPI4_SCK, PIN_SPI4_MOSI, PAD_SPI4_TX, PAD_SPI4_RX);
#endif
#if SPI_INTERFACES_COUNT > 5
  SPIClass SPI5(&PERIPH_SPI5, PIN_SPI5_MISO, PIN_SPI5_SCK, PIN_SPI5_MOSI, PAD_SPI5_TX, PAD_SPI5_RX);
#endif
```

SapientHetero, 2018-08-12:
I got the SdFat version of the "bench" example sketch working on the Metro M4 today by reducing the SPI clock setting to 12 MHz from the default of 50 MHz. At higher settings the SdFat library was unable to initialize the card.

Throughput on the Metro M4 at 4, 8 and 12 MHz was higher than the best I got on the Mega or Due. (though only slightly faster than the Due). I've posted them below for posterity. All tests were run with the Adafruit WINC1500 shield and a SanDisk Ultra UHS-I Class 10 32GB micro SD card.

**Adafruit Metro M4 with SPI clock set to 12 MHZ**

Use a freshly formatted SD for best performance.

Type any character to start
FreeStack: 127084
Type is FAT32
Card size: 31.91 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: SC32G
Version: 8.0
Serial number: 0X111A0968
Manufacturing date: 2/2012

File size 5 MB
Buffer size 65536 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
350.14,464142,179390,187113

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
683.23,96923,95442,95916

Done 

**My DUE bench sketch results (SdFat 1.0.6) at 40 MHz with a 65536 byte buffer were:**

Use a freshly formatted SD for best performance.

Type any character to start
FreeStack: 26548
Type is FAT32
Card size: 31.91 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: SC32G
Version: 8.0
Serial number: 0X111A0968
Manufacturing date: 2/2012

File size 5 MB
Buffer size 65536 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
621.66,131325,103974,105336
623.84,118452,103765,104967

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
654.93,101129,99958,100060
655.02,101035,99959,100058

Done

**MEGA 2560 performance**
Use a freshly formatted SD for best performance.

Type any character to start
FreeStack: 6711
Type is FAT32
Card size: 31.91 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: SC32G
Version: 8.0
Serial number: 0X111A0968
Manufacturing date: 2/2012

File size 5 MB
Buffer size 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
281.23,27624,1572,1813
284.88,95152,1540,1790

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
507.68,3108,988,1001
507.84,1992,988,1001

SapientHetero, 2018-08-17:
As noted previously I got my Class 10 SD card to work with the Metro M4 by slowing the clock down to 12 MHz. But this week I integrated the Seeed Ethernet 5500 shield with my application for wired Ethernet, and it reportedly works only with cards up to 2GB in size. The only 2GB cards I've found are much slower than Class 10. I can initialize a 2GB card with the Seeed shield mounted on an Arduino Mega with the SPI clock reduced to 4MHz, but haven't been able to initialize it on a Metro M4 at any clock speed.

The sketch used in that test works with the same 2GB SD card using an Adafruit WINC1500 shield and Metro M4, with SPI clock set to 12 MHz. I'm guessing that it's an issue with either the Seeed shield or its library.

Ryan, I tried your SPI lib version but it didn't help either. I reduced the SPI clock speed down to as little as 500 Hz [SD_SCK_HZ(500)] but it just took longer to time out.

ladyada, 2019-05-24:
we recently forked and updated sdfat to add SPI DMA and it runs at the theoretical-max of 3MBps
https://github.com/adafruit/SdFat
closing as its resolved!

closed 2019-05-24 by ladyada