adafruit/ArduinoCore-samd #59: UART has no transmit buffer.

WestfW, 2018-08-10:
The SAMD21 code is currently using a busy-loop for writing to UART devices.  I consider that moderately unacceptable, especially on chips like the M4 which have lots of RAM.
The Arduino core added txbuffer support in July 2017; it looks like the Adafruit branch has not picked up that change.  (You could do better than they did: I'm not thrilled that the Arduino core forces the use of the same-sized buffer for transmit and receive.)
https://github.com/arduino/ArduinoCore-samd/commit/b24c9e5b6851ba70bc4bfa6b454f3a999d50262c#diff-9aa55b794319d1ec143466ac42b86e54

stingEM, 2018-08-16:
Hey, does anyone have two minutes to look at this error/issue? Noticed no one is assigned yet. 

Would help a lot if this got fixed. Thanks! 👍

ladyada, 2018-08-17:
ha! its a lot more than 2 minutes to look at - we have to do a rebase and pick up all the arduino commits, and theres maybe 100 of them. we'll try to do in the next month or so since its been on our list!

ladyada, 2018-09-05:
we've merged with upstream - so this should be fixed if they fixed it in arduino's samd! will close, re-open if ya find it didn't merge right :)

closed 2018-09-05 by ladyada

WestfW, 2018-09-05:
That seems to have gotten it...
https://github.com/adafruit/ArduinoCore-samd/blob/master/cores/arduino/Uart.cpp#L185

ladyada, 2018-09-05:
sweet!