adafruit/ArduinoCore-samd #62: attachInterrupt(11) clobbers previous attachInterrupt()

drewfish, 2018-10-08:
Starting with version 1.2.3 when I `attachInterrupt(11, cb11)` whatever my interrupt I registered for the previous interrupt is not called and instead the interrupt for 11 is called.

This is my minimum reproducing case. You'll need to wire 10 & 11 through buttons to ground. It prints `pin 11` for either button.

```cpp

uint16_t lastPin = 0;

void EICPin10() {
    lastPin = 10;
}

void EICPin11() {
    lastPin = 11;
}

void setup() {
    pinMode(10, INPUT_PULLUP);
    pinMode(11, INPUT_PULLUP);
    pinMode(12, INPUT_PULLUP);
    attachInterrupt(10, EICPin10, RISING);
    attachInterrupt(11, EICPin11, RISING);
}

void loop() {
    if (Serial && lastPin) {
        Serial.print("pin ");
        Serial.println(lastPin);
        lastPin = 0;
    }   
}
```

I did not see this behavior in 1.2.2. (My workaround is to downgrade to that version.)

drewfish, 2018-10-08:
Oh I forgot to say I'm using a Feather M0 Adalogger.

deanm1278, 2018-11-05:
Hey I have fixed interrupt handling in this commit: https://github.com/adafruit/ArduinoCore-samd/commit/0654cd01b3e45c3c10c7674daf4b47ba46b80052

closed 2018-11-05 by deanm1278

drewfish, 2018-11-06:
Yep, looks like it's fixed in version 1.2.6. Thanks!

drewfish, 2018-11-06:
It doesn't appear that this fix is in the platformio.org platform "atmelsam" yet (3.4.1). They probably just need to kick off their builds in order to pull in the latest of this repo.

deladriere, 2018-12-22:
@drewfish Same problem with the Feather M0: I have asked PIO to update their SAM core
Thanks for the workaround!