adafruit/ArduinoCore-samd #65: 'RingBufferN' does not name a type

lsh940131, 2018-11-05:
I have DC Motor + Stepper FeatherWing and Adafruit M0,
So I test motor using Adafruit_MotorShield library,
but <Wire.h> library occurs some trouble.

there is in the Wire.h library code
#include "RingBuffer.h"
RingBufferN<256> rxBuffer;

of course I have RingBuffer.h library. However, the following error occurs.
C:\Users\CoTLab\Documents\Arduino\libraries\Wire/Wire.h:74:5: error: 'RingBufferN' does not name a type
     RingBufferN<256> rxBuffer;

Why are you doing this?

ladyada, 2018-11-05:
check you have the latest board support packages - that you are selecting the right board in tools menu, etc....we've never heard of this issue!

ladyada, 2019-03-09:
closing cause no updates and i think its your custom Wire library?

closed 2019-03-09 by ladyada