adafruit/ArduinoCore-samd #75: pulse_asm.S not conditionalized for CM4 (samd51)

WestfW, 2019-01-08:
pulse_asm.S was produced by compiling C code for SAMD21, and still has assembler directives for that CPU, causing it to produce a .o file that the gnu linker does NOT want to link into M4 binaries:
```
	.cpu cortex-m0plus
	.fpu softvfp
	.eabi_attribute 20, 1	@ Tag_ABI_FP_denormal
	.eabi_attribute 21, 1	@ Tag_ABI_FP_exceptions
   eabi_attribute 23, 3	@ Tag_ABI_FP_number_model
```

The floating point stuff seems to be more of an issue that the CPU defintions (m4 being a proper superset of M0+) - the actual error is something like:
```
arm-none-eabi/bin/ld: error: sketch_jan07a.ino.elf uses VFP register arguments, arduino_build_92936/core/core.a(pulse_asm.S.o) does not
arm-none-eabi/bin/ld: failed to merge target specific data of file /var/foldersarduino_build_92936 core/core.a(pulse_asm.S.o)
```

I succeeded in getting a test sketch to compile by just deleting those lines - should cpu and fpu be command-line options?  (But ... there's apparently desire to keep the .S close to what the compiler produces ?  So I'm not sure how to actually proceed, patch-wise.)

Customer report: https://forums.adafruit.com/viewtopic.php?f=63&t=145803

oesterle, 2019-01-08:
It also might be valuable to look at Paul Stoffregen's implementation of `pulseIn()` for M4 boards like the Teensy 3.0 and later. That version of `pulseIn()` doesn't use assembly language, which also makes the implementation more portable.

https://github.com/PaulStoffregen/cores/blob/master/teensy3/pins_teensy.c

WestfW, 2019-01-08:
I think it needs a sort of "executive decision" on "direction."
I don't really understand why this is a .S file in the first place, especially on the faster ARM chips.  It doesn't seem to have been particularly optimized (although I guess that having the timing NOT change from release to release of the compiler might be useful?)
Also, I'm not very comfortable with anything using a counting loop, in the presence of cache/etc. :-(
Perhaps, for the zippy m4s, simpler code in pulse.c could be used to eliminate pulse_asm.S all together...  Let me see if I can put together a us-scale pulser and do some testing...

ladyada, 2019-01-08:
i'd skip the assembly code, and go straight to C, its not going to be much slower!

WestfW, 2019-01-09:
re-wrote pulseIn(), doing the timing with micros()
It seems to work; I tested values between 0 and about 1.2s with accuracy ~+/-1 us (Test program (Attached) is limited to 1.3s pulses, approximately.)

"parser" is at https://github.com/WestfW/parser (somewhat embarrassingly not C++-ish)
[pulseGen.zip](https://github.com/adafruit/ArduinoCore-samd/files/2740130/pulseGen.zip)

ladyada, 2019-01-09:
thanks! merged in, we'll do a release soon :)

closed 2019-01-09 by ladyada