adafruit/ArduinoCore-samd #87: analogWrite() oddness on M4 when using both DACs

caternuson, 2019-02-12:
Reference:
https://forums.adafruit.com/viewtopic.php?f=62&t=146612&p=727668#p727280

Modified example taken from above.
```cpp
void setup() {
  analogReadResolution(12);
  analogWriteResolution(12);
  analogWrite(A0,2000);
  delay(1000);
}

void loop() {
  int a2count = analogRead(A2);
  delay(50);
  analogWrite(A1,a2count);
}
```

Confirmed odd behavior with above simple example, but reduced initial delay in setup() to 1 sec.
[image]

ladyada, 2019-02-12:
oi - replied and asked if they could kindly try earlier BSPs, perhaps it was introduced at some point

caternuson, 2019-02-12:
Something about the first call. Behavior doesn't repeat with subsequent calls. This:
```cpp
void setup() {
  analogWriteResolution(12);
  analogWrite(A0,500);
  delay(1000);
  analogWrite(A1,500);
  delay(1000);
  analogWrite(A0,1000);
  delay(1000);
  analogWrite(A1,1000);
  delay(1000);
  analogWrite(A0,2000);
  delay(1000);
  analogWrite(A1,2000);
  delay(1000);
  analogWrite(A0,3000);
  delay(1000);
  analogWrite(A1,3000);
  delay(1000);
}

void loop() {
}
```
Does this:
[image]

Something here?
https://github.com/adafruit/ArduinoCore-samd/blob/master/cores/arduino/wiring_analog.c#L428

kevinjwalters, 2019-07-03:
Not related to using both together but I've found another peculiarity of the SAMD51 DACs, see https://forums.adafruit.com/viewtopic.php?f=24&t=153707 . I have demonstrated it with CircuitPython.

kevinjwalters, 2019-07-08:
The DACs have a "differential" mode so that feature could lead to some linkage between the two DACs? I'm not saying it's in differential mode, btw, could just be a bug somewhere relating to this, possibly the hardware.

It looks like there's a tiny burst of noise on `DAC1` too for no reason, appears just under bottom right of `1` in `DAC1`? Does that coincide with "upward" noise on `DAC0` or does the printing of DAC0 have a black background that's blotting out the upper part of noise?

To explore phenomena, might be worth trying:
1. changing the order of DAC0/DAC1 changes,
2. starting high and then using descending values,
3. change timings to ensure timing is not coincidence,
4. add an initial delay of 17 seconds to rule out any other things (?) initialising near program start-time affecting it.

kevinjwalters, 2019-07-09:
In CircuitPython the DAC output is weird just from initialisation, e.g. try this:

```
import board
import audioio
dacs = audioio.AudioOut(board.A0, right_channel=board.A1)
```