adafruit/ArduinoCore-samd #88: Some mistakes (probably) with the Clocks on SAMD51

aryelevin, 2019-02-12:
Hi,

I've found two bugs on the startup.c code in the SAMD51 variant.

First one is that after initialising the GCLK7, the code wait for syncing with the GCLK5 (the older clock generator).
BTW, why did this changed? (from 5 to 7)
The second one is that for GCLK4, the GCLK_GENCTRL_DIVSEL flag makes it in fact to be 1.5MHz instead the expected 12MHz.

In addition, the same issue is with the GCLK7 (previously was 5), it divided by 24 which makes it to be 2MHz instead the expected 1MHz.

Hope I helped.

Best

ladyada, 2019-02-12:
hiya, i'm not sure why the clocks changed :/ but you can submit a PR and we'll try it out!

aryelevin, 2019-02-12:
I'm sorry, for some reason I don't have the repo checked out with git.
I can post my fixed file if that acceptable for you.

In addition, its weird, since somewhere there it is respected as GCLK7, and looks like it changes by intention.
I'm not sure.

I discovered the issues with the dividers by configuring everything on the code flow on the start.atmel.com app, and I saw the issues with the frequencies.

aryelevin, 2019-02-12:
It seems to be that it changed on this commit:
https://github.com/adafruit/ArduinoCore-samd/commit/982819103040176bc8ab2e5a6e99a75e1560e602
Something to do with grand central?

ladyada, 2019-02-12:
im not sure, and the person who wrote that code isn't available to answer Q's so i think yeah if you can submit a PR to try fixing that clock - i can try it!

aryelevin, 2019-02-12:
OK, I will

sajattack, 2019-02-13:
:+1: I found this same issue 2 hours after you did lol.

ladyada, 2019-02-13:
@sajattack yay! i think :) wanna check arye's PR when its ready? or help author if ya like

sajattack, 2019-02-13:
Sure thing.

aryelevin, 2019-02-13:
Hi @ladyada and @sajattack , 

Added PR https://github.com/adafruit/ArduinoCore-samd/pull/89

Waiting for your input about it.

closed 2019-02-13 by ladyada