adafruit/ArduinoCore-samd #90: bossac not found under Linux

Wene, 2019-02-17:
When trying to upload a sketch under Linux, the following error occurs:

> java.io.IOException: Cannot run program "{runtime.tools.bossac-1.7.0.path}/bossac": error=2, Datei oder Verzeichnis nicht gefunden

I could quickly fix it by manually changing the path in platform.txt line 115 to the absolute path of the tools- folder in the package and the command in line 116 from "bossac" to "bossac_linux" . But I think there will be a more elegant solution with relative paths.

ladyada, 2019-02-17:
please make sure you have all the latest SAM packages from arduino and adafruit

Wene, 2019-02-19:
> please make sure you have all the latest SAM packages from arduino and adafruit

Thanks for the hint. After installing the "Arduino SAMD Boards" it works with the original config.

In this case I wonder why the "bossac_linux"- tool is included here since it's not supposed to be used. And of course I would appreciate it if this hint could be added to the README.

ladyada, 2019-02-19:
its a bit of an ancient remainder - we should delete it

Wene, 2019-04-01:
Has been solved with the current version. The misleading tools are removed in the library.

closed 2019-04-01 by Wene