adafruit/ArduinoCore-samd #92: Update Ringbuffer.h

lunng, 2019-03-07:
RingBuffer.h in the adafruit defines the buffer size as 164 bytes, please add #ifdef wrapper to this statement to resemble the arduino core samd github. This will allow for extension of the buffer without having to edit the library file. Thank you.

ladyada, 2019-03-07:
this might be a better one to submit upstream to arduino?

lunng, 2019-03-07:
The arduino upstream does contain this line, I don't know why the changes aren't synced

lunng, 2019-03-07:
see https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/RingBuffer.h

ladyada, 2019-03-08:
we haven't done a upstream sync lately - we should do that soon!

ladyada, 2019-03-08:
ok check the latest in master (not in board support package release yet!) 
re-open if still an issue :)

closed 2019-03-08 by ladyada