adafruit/ArduinoCore-samd #94: analogWrite() on SAMD51 DAC has 10ms internal delay?

WestfW, 2019-03-12:
AnalogWrite() to the DACs on SAMD51 has an explicit call to delay:

https://github.com/adafruit/ArduinoCore-samd/blob/master/cores/arduino/wiring_analog.c#L453
~~~
				else if(channel == 1){
					while ( !DAC->STATUS.bit.READY1 );
					while (DAC->SYNCBUSY.bit.DATA1);
					DAC->DATA[1].reg = value;
				}
				delayMicroseconds(10000);
}
~~~

That's a really long delay for what ought to be a "non-blocking" call to a 1MHz DAC peripheral!
Is it left over from testing, or something?

dhalbert, 2019-03-12:
It was this commit:https://github.com/adafruit/ArduinoCore-samd/commit/540780d3d14859207b7da43c3c5545ac89a86fff. It changed from `delay(10)` to `delayMicroseconds(10000)`. I don't have a reason why it's there, but it must have been deliberate due to the change in delay calls.

ladyada, 2019-03-12:
@WestfW its only on the first write to the DAC, right?

WestfW, 2019-03-14:
> only on the first write

Oh.  Right.   I missed that.  Nevermind.
(analogWrite() still seems much slower than it ought to be, but...  Investigating further.)

closed 2019-03-14 by WestfW

WestfW, 2019-03-15:
Ahh.  GENCLK4 was not configured correctly.  It had DIVSEL set, causing it to be 48MHz/(2**4) instead of 48MHz/4.

This is already fixed in the source, by https://github.com/adafruit/ArduinoCore-samd/pull/89/files
But I guess it hasn't made it to a release yet.

ladyada, 2019-03-15:
hiya it should be in a release - we did one a couple days ago - please check & let us know :)

WestfW, 2019-03-15:
(yep.  Confirmed that it's better in the new release.  I guess I was expecting the IDE to prompt me to update when new releases were available...)
The analogWrite() version now gets about 200Hz; very close to expected speed, though I'm not sure that it's doing thing "right" - I have feelers out to better explain the DAC...

kevinjwalters, 2019-07-08:
@WestfW Did you mean 200kHz? I'm using CircuitPython and the audio library doesn't seem to change the output when I go beyond 200k. Discussed here although there's not much more info than this at the moment: https://forums.adafruit.com/viewtopic.php?f=60&t=153797

There's another oddity in how the DAC rises, it can be very steppy. That depends on timing and doesn't affect falls: https://forums.adafruit.com/viewtopic.php?f=24&t=153707