adafruit/ArduinoCore-samd #95: Serial.read no longer working

freemovers, 2019-03-22:
I noticed that the Serial.read is no longer works on this latest version. Used the example SerialEvent to verify this issue. There is no response, and the RX light is no longer flashing after sending data over serial to the board. Sending from the board still works fine. Tested this on the Metro M4 and Feather M4.

ladyada, 2019-03-23:
@freemovers thats exciting, can you regression test and let us know when it stopped working? that will help us narrow it down

freemovers, 2019-04-04:
Looks like to code never gets to `usb.send` on line#221. `_usbLineInfo.lineState` always returns 0. The following comment is shown there as well: "// Problem with Windows(R)"
I am not to familiar with the high-level CDC. I am just setting up a connection serial connection in C# in Visual Studio, running Windows 10.

ladyada, 2019-04-04:
sorry cant reproduce - this code works just fine on our windows 10

```
int led = 13;

void setup() {
  // initialize the digital pin as an output.
  pinMode(led, OUTPUT);     
  Serial.begin(115200);
  while (!Serial);
  delay(100);
  Serial.println("USB Serial Echo Test");  
}

// the loop routine runs over and over again forever:
void loop() {
  if (Serial.available()) {
    digitalWrite(led, HIGH);
    Serial.write(Serial.read());
    digitalWrite(led, LOW);
  }
}
```

freemovers, 2019-04-04:
Sorry about that, should have tested it from the serial monitoring as well. That seems to work just fine. I get the actual response from Realterm as well from my sketch (sending hex values), but not from any .net application. Thank you for all the support!

closed 2019-04-04 by freemovers

frejanordsiek, 2019-09-23:
Bringing this back up because I am having the same issue on a Grand Central (in addition to having a hard time uploading sketches larger than tiny, but that is a separate problem with a work around) but can't get it to work regardless of what application I use to connect to the serial on my computer.

I am using the Arduino example

04.Communication : SerialEvent

I have tried interfacing the serial from Arduino's serial monitor, CuteCom, and using the pyserial package in Python.

My OS is Ubuntu 19.04 x86-64.
My Arduino IDE is 1.6.10 using the latest Adafruit SAMD Boards package from the board manager (1.5.3).

There is one potential clue that I have noticed which might be of help (or might not be). After writing 12 bytes to serial from my computer, it hangs on the 13th byte. I don't know if this is a UART buffer issue on my computer or the microcontroller, or something else. But it might be a clue.

ladyada, 2019-09-23:
hiya this sounds like a tech support issue, please post to the adafruit forums :)

frejanordsiek, 2019-09-23:
OK. I will do so. Hopefully it is just a tech issue.