adafruit/ArduinoCore-samd #96: Bugs in SAMD51 Audio Library

ianm8, 2019-04-01:
There are a few bugs in the Audio library that prevent some of the objects working and ADC1 in the AudioInputAnalogStereo object.

The audio library is here: [](https://github.com/adafruit/Audio)
Original discussion is here: [](https://forums.adafruit.com/viewtopic.php?f=62&t=148013&p=739669)

Further details below:

ladyada, 2019-04-02:
sweet, can you submit a PR with the fixes ?

ianm8, 2019-04-02:
There are various files that check for the definition of KINETISK. Without it the library does not produce correct code for the SAMD51. In some instances there is a check for __SAMD51__ as follows:
#if defined(KINETISK) || defined(__SAMD51__)

I used a temporary fix by adding #define KINETISK to Audio.h.

ladyada, 2019-04-02:
ahh i'd rather we do it right, could you try to submit a PR? i'd appreciate it a ton - especially if you can test it

ianm8, 2019-04-02:
OK, it'll take a while! :)

ladyada, 2019-04-02:
ok you can also check if other people in that thread want to help!

ianm8, 2019-04-02:
I'll make a start and see how it goes.

ianm8, 2019-04-05:
OK, I've made the changes to a GitHub Desktop pull. Now how do I do a PR?

Cheers, Ian

ladyada, 2019-04-05:
see
https://learn.adafruit.com/an-introduction-to-collaborating-with-version-control
and
https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github