adafruit/ArduinoCore-samd #97: SAMD51 PWM analog write is only 8 bit

jepye, 2019-04-09:
Why is the SAMD51 set to use only 8 bit for both the TC (via COUNT8) and TCC (via 1/256 prescaler) timer instances of PWM in analog write?

https://github.com/adafruit/ArduinoCore-samd/blob/904acd10715962899698500e3bfdc7581c4f99c8/cores/arduino/wiring_analog.c#L510-L552

ladyada, 2019-04-09:
i think this is for the grand central so we can support all the PWM outputs on pins. its been a while tho :)

jepye, 2019-04-09:
I switched it to 16 bit locally because we needed it a while back, but I
think it may have caused some other issues, but can't remember.  I'll try
to find the file.

GMagician, 2019-06-14:
I think 8 bit has been chosen because of buffered option. I'm wondering why not to use 16 bits and downslope?