adafruit/ArduinoCore-samd #99: Feather M0 Express variant.h SPI1 defintion

mattracing, 2019-04-17:
"
#define PIN_SPI1_MISO         (36u)
#define PIN_SPI1_MOSI         (37u)
#define PIN_SPI1_SCK          (38u)
#define PERIPH_SPI1           sercom2
#define PAD_SPI1_TX           SPI_PAD_0_SCK_1
#define PAD_SPI1_RX           SERCOM_RX_PAD_2

static const uint8_t SS1   = 39 ;	// HW SS isn't used. Set here only for reference.
static const uint8_t MOSI1 = PIN_SPI_MOSI ; **<---PIN_SPI1_MOSI???**
static const uint8_t MISO1 = PIN_SPI_MISO ; **<---PIN_SPI1_MISO???**
static const uint8_t SCK1  = PIN_SPI_SCK ; **<---PIN_SPI1_SCK???**
"

ladyada, 2019-04-17:
woops thats a typo, but it isnt used anyhow cause we talk to the QSPI w/SPI1 :D

mattracing, 2019-04-17:
Okey. ;-)

closed 2019-04-22 by ladyada