adafruit/ArduinoCore-samd #113: Feather M0 Bluefruit missing from boards.txt

jeff-winn, 2019-05-13:
The Feather M0 Bluefruit board ([product link](https://www.adafruit.com/product/2995)) is missing from the boards.txt file with the VID and PID settings causing issues with board detection within the Arduino CLI.

- vid "0x239A"
- pid "0x0015"

Possible fix would be adding the following settings to the boards.txt file within the Adafruit Feather M0 (SAMD21) section:

> adafruit_feather_m0.vid.2=0x239A
> adafruit_feather_m0.pid.2=0x0015

I'm not entirely sure whether this is the correct patch, or if another board should be added with the configuration duplicated from the Feather M0 board configuration.

ladyada, 2019-05-13:
yeah thats a good idea! i just added, want to try master branch and/or just adding that yourself. it'll get picked up in the next release

jeff-winn, 2019-05-14:
I already made the changes locally to ensure everything was being detected correctly prior to creating the issue and it's been happy since I made the change.

Thanks for putting the patch in, and love your feather products!

ladyada, 2019-05-14:
yayyyy! closin'

closed 2019-05-14 by ladyada