adafruit/ArduinoCore-samd #126: samd51..missing calibration defines

GMagician, 2019-06-07:
Samd51 has a special reserved are with calibration values, its address is 0x00800080.

> The NVM Software Calibration Area contains calibration data that are determined and written during production test. These calibration values should be read by the application software and written back to the corresponding register.

In this area there are some ADC calibration values that "software" has to load (ADCx BIASCOMP, ADCx BIASREFBUF, ADCx BIASR2R, ...).
Since there is no reference to such memory address in this core then I think that these values are never loaded by anyone...

Is not a good solution to add defines for this area?

ladyada, 2019-06-07:
sure - please submit a PR for these defines!

GMagician, 2019-06-07:
I can do that but where to put file? which name to give him?
File hierarchy?

ladyada, 2019-06-07:
could put it in the analog header file or the variant?

commit 6aa6dc17 (2016-05-14): Added definitions for DAC0 in variants

Fix #126
closed 2019-06-08 by GMagician