adafruit/ArduinoCore-samd #128: SAMD51 Voltage Regulator Default

jacky4566, 2019-06-11:
In pursuit of maximum power savings I noticed that Adafruit is not using the onboard SAMD51 Buck regulator despite the inclusion of an inductor onboard. Is there any reason for this?
On my own custom board I brought the idle from 10.4mA to 6.1mA just with one line.
`SUPC->VREG.bit.SEL = 1;`

https://github.com/adafruit/ArduinoCore-samd/blob/cec0a6bd7d69d7eb5bacf7c33e6f0985046d186b/cores/arduino/startup.c#L247

ladyada, 2019-06-11:
you can totally use it if you like, but it makes the ADC noisy so we haven't used it much.

ladyada, 2019-06-11:
added a note here on using it. https://learn.adafruit.com/adafruit-metro-m4-express-featuring-atsamd51/adapting-sketches-to-m0#enabling-the-buck-converter-on-some-m4-boards-7-27

closed 2019-06-11 by ladyada

jacky4566, 2019-06-11:
Whoah! @ladyada had her coffee this morning! Makes sense about the ADC.
Thanks that'll probably do it then!

ladyada, 2019-06-11:
🍵 😂