adafruit/ArduinoCore-samd #133: repureposing SERCOMs

WestfW, 2019-06-26:
It would be "nice" if there was an easier way to assign functionality (UART/I2C/etc) to the unused SERCOMs on a board.  Or even to repurpose in-use SERCOMs to some other function.  (for example, the Uno3-style I2C pins could become an additional UART.)

Especially on the SAMD51 boards.

I don't know offhand if this is practical...

ladyada, 2019-06-26:
its pretty fast to follow this guide - we do it all the time - do you want to write a helper?
https://learn.adafruit.com/using-atsamd21-sercom-to-add-more-spi-i2c-serial-ports

freemovers, 2019-06-26:
The guide is great and helped me a lot, but I understand it could be complicated to determine which pins to use for each SERCOM combination. For that reason I use Atmel Start to make sure the selected combination is possible for the required protocol.

s-light, 2019-08-13:
i think the tutorial is a very good resource- 
but would be good to add a note that for SAMD51 there are differences..
for SPI SCK/MOSI you have fewer options for the PAD selections:
[SERCOM SPI  - 35.8.1 Control A -- Bits 17:16 – DOPO[1:0] Data Out Pinout](http://ww1.microchip.com/downloads/en/DeviceDoc/60001507E.pdf#page=986&zoom=auto,-110,319)
SCK is only possible on PAD1
(in the linked file its on page 986)
in encounterd this as i tried to follow the guide..

nekuneko, 2019-08-31:
Hello everyone, 
these are all SERCOM differences in SAMD51, it would be nice to update the tutorial with them, because they are really are error prone if you design custom boards (changes are in **bold**): 

SAMD51 SERCOM RULES:

* **UART**: (PAGES #945, 946 CRTLA A, BITS RXPO[21:20], TXPO[17:16])
    -  RX: 0, 1, 2, 3
    - **TX: 0**

* **SPI**: (PAGE #986 CRTLA A, BITS DIPO[21:20], DOPO[17:16]):
    - RX_MISO: 0, 1, **2**, 3
    - **TX_MOSI_0_SCK_1**
    - **TX_MOSI_3_SCK_1**

* **I2C**: (PAGE #1006, Section 36.4 Signal Description)
    - SDA: 0
    - SCL: 1

Refer this datasheet: http://ww1.microchip.com/downloads/en/DeviceDoc/60001507E.pdf.
Thank you very much for your tutorials LadyAda!

ladyada, 2019-09-02:
OK i added sections to each page. I will close this now, it should be resolved

closed 2019-09-02 by ladyada

WestfW, 2019-11-06:
I was thinking of something more dynamic and run-time based.

sercom.cpp already has nice SERCOM::initUART, SERCOM::initSPI, and etc that look close to being able to change the behavior of a sercom at any time, and it looks like the main stumbling block is that the individual UART/SPI/Wire modules tie ISR handlers to the function at compiler time.  The SAM chip vector tables COULD be relocated to RAM instead of flash, which would enabled changing the ISRs for each sercom at runtime as well.

I think that relocating the vectors to RAM, although it sounds like a major change, would actually be relatively easy and low-risk, and might have other advantages as well.

I'll try to come up with some example code "soon."  (sigh.)