adafruit/ArduinoCore-samd #139: Feather board m4 express: How do i get 1 MSPS on the ADC? (Arduino IDE)

koen1993, 2019-07-09:
I wanted to get 1 MSPS on the adc, or as close to as possible. With my current configuration i only reach 140k values/s. What registers do i need to configure? Did i miss something? My code below

```
#define ADCX ADC0

long curr_time = millis();
uint32_t sample_count = 0;

void ADC_Init(void)
{
   //////////////////////////////////////////////////////////////////////////
   ///ADC Clock Config
   //////////////////////////////////////////////////////////////////////////
   //GCLK->CTRLA.bit.SWRST = 1;
   //while(GCLK->SYNCBUSY.bit.SWRST);
   MCLK->APBDMASK.bit.ADC1_ = 1;
   MCLK->CPUDIV.reg = 0x01; //Div 1
   GCLK->PCHCTRL[ADC0_GCLK_ID].reg = GCLK_PCHCTRL_GEN_GCLK1_Val | (1 << GCLK_PCHCTRL_CHEN_Pos); // enable gen clock 1 as source for ADC channel
   GCLK->GENCTRL[0].reg = GCLK_GENCTRL_SRC_DFLL | GCLK_GENCTRL_GENEN | GCLK_GENCTRL_DIV(1);
   GCLK->GENCTRL[0].bit.DIVSEL = 0;
   GCLK->GENCTRL[0].bit.DIV = 0;

  
  ADCX->CTRLA.bit.PRESCALER = ADC_CTRLA_PRESCALER_DIV2_Val;
  
  //Choose 12-bit resolution
  ADCX->CTRLB.bit.RESSEL = ADC_CTRLB_RESSEL_8BIT_Val;
  ADCX->CTRLB.bit.FREERUN = 1;
  

  while(ADCX->SYNCBUSY.reg & ADC_SYNCBUSY_CTRLB); //Potentiel endlosloop: Beobachten
  

  //Sampletime
  ADCX->SAMPCTRL.reg = 0;

  while(ADCX->SYNCBUSY.reg & ADC_SYNCBUSY_SAMPCTRL);

  ADCX->INPUTCTRL.reg = ADC_INPUTCTRL_MUXNEG_GND;   // No Negative input (Internal Ground)

  while( ADCX->SYNCBUSY.reg & ADC_SYNCBUSY_INPUTCTRL );  //wait for sync
  
  ///only 1 sample
  //ADCX->AVGCTRL.bit.SAMPLENUM = 0;
  ADCX->AVGCTRL.reg = 0;

  while(ADCX->SYNCBUSY.reg & ADC_SYNCBUSY_AVGCTRL);

  
  //Select VDDANA (3.3V chip supply voltage as reference)
  ADCX->REFCTRL.reg = ADC_REFCTRL_REFSEL_INTVCC1;   

  //Enable ADC
  ADCX->SWTRIG.bit.START = 1;
  ADCX->CTRLA.bit.ENABLE = 1;
  
  //wait for ADC to be ready
  while(ADCX->SYNCBUSY.bit.ENABLE);   
}

void setup() 
{
  ADC_Init();
  Serial.begin(115200);
  while(!Serial);
  Serial.println("Begin ...");
} 

void loop() 
{
  //ADC0->SWTRIG.bit.START = 1;
  //while(!ADC0->INTFLAG.bit.RESRDY);
  //long int  one_sample = ADC0->RESULT.reg;
  analogRead(ADC_PIN);
  sample_count++;
  if(((curr_time / 1000) + 1) <= (millis()/1000))
  {
    Serial.println(sample_count);
    sample_count = 0;
    curr_time = millis();
  }
}
```

ladyada, 2019-07-09:
we don't have any example code, you may find the best place to get that kind of advice/assistance is from Microchip's engineers, they would have example code in ASF that you could try and then back-adapt

ladyada, 2019-07-09:
also check the eratta, 500ksps may be the max

koen1993, 2019-07-09:
Allright, yeah it says that ADC TUE/INL/DNL is not guranteed, when sampling frequency is above 500ksps (or ADC Vref is different from VDDANA). But i wanted to see how bad it is anyway.

I also figured it might be that the XSOC clock need to be configured for exact 32MHz DIV2. In my example it run with 48MHz DIV4: This should theoreticaly result in a samplefrequency of 0.75MSPS (Which, as mentiond above, is only around 460ksps).

Anyway, thanks for your feedback, i hoped you had something and i have also contacted Microship via the support.

PS: I will close this issue for now, if i do get a solution and i am still able to edit the post, i will edit it into my last post. (i am relativly new to github and don't know if i am able to edit an closed thread)

closed 2019-07-09 by koen1993

ladyada, 2019-07-09:
ok if you get something working you can submit a PR with an example sketch, i think that could be handy