adafruit/ArduinoCore-samd #141: Add bossac and other build tools to the core

PTS93, 2019-07-18:
Not sure if a policy from Arduino is preventing you from doing this but it would be great if the tools that are normally installed by the Arduino provided SAMD core could be installed by the Adafruit core as well (or doing it only if the Arduino SAMD core is not installed if its possible to check that).

I had numerous complaints from users who only installed the Adafruit core in believe that this will install everything that is needed. It is indeed quite unintuitive that you would need the other core as well just for the build tools. I had it happen to myself quite a few times and its an issue that is not very transparent to new users that they need to install another seemingly unrelated package.

Maybe its also a deeper issue that the Board Manager does not have a system for dependencies to other packages, which would solve this issue easily.

ladyada, 2019-07-18:
we dont think there's any policy, we dont know how to force-install those tools. if you have any suggestions, we'd prefer to have a stand-alone core :)

PTS93, 2019-07-18:
Huh, maybe this was resolved a while ago and I just didn't notice. I just removed both arduino and adafruit folder from my packages and installing just the latest adafruit core created also a new arduino folder with just the tools inside. Though I just tested this on Ubuntu right now, the last time I had the issue with a colleague was on a windows machine.
I will test again on Windows 10 on my home machine to see if it behaves any differently.
The most recent issue was today when I discovered https://github.com/adafruit/ArduinoCore-samd/issues/142 so maybe there is just something in the new 'official' core that messes with third party cores and this issue here doesn't technically exist anymore.

ladyada, 2019-07-18:
ok! its tough to figure out sometimes :) plz close this issue if its resolved

PTS93, 2019-07-18:
Will do once I verified it on windows

PTS93, 2019-07-18:
Ok working on my Windows 10 machine as well after making sure that I deleted both package folders and the staging folder from `Arduino15` and freshly downloading Adafruit 1.5.2.
I have to assume that issues before were just before the dependency was added or because of older Arduino installation with a lot of package baggage where something got messed up.

closed 2019-07-18 by PTS93