adafruit/ArduinoCore-samd #142: Upload failing with newest Arduino-SAMD core due to missing bossac

PTS93, 2019-07-18:
Installing the latest Arduino SAMD core (1.8.1) under Windows 10 causes the Adafruit core to not find Bossac. I'm not exactly sure what the exact cause is but `{runtime.tools.bossac-1.7.0.path}` does not seem to resolve to bossac anymore. It is fixed by installing Core version 1.6.x

This is the exact error message that was reproduced on two seperate Windows 10 machines.
`java.io.IOException: Cannot run program "{runtime.tools.bossac-1.7.0.path}/bossac": CreateProcess error=2,`

ladyada, 2019-07-18:
odd - we have it down as a tool dependency
https://github.com/adafruit/arduino-board-index/blob/gh-pages/package_adafruit_index.json#L3674
if you can help us figure out what we're missing, that would be great - there's not a lot of documentation about it

PTS93, 2019-07-18:
Do you know where this `runtime.tools.bossac-1.7.0.path` is set? I could not figure out how the path is determined, this is likely the root cause of the issue.

ladyada, 2019-07-18:
no idea, that's something maybe @cmaglie or @sandeepmistry may know - i assume it somewhere within the arduino IDE

ladyada, 2019-12-31:
@PTS93 is this resolved

PTS93, 2019-12-31:
I have seen this again after uninstalling and reinstalling the core but I think this is not fixiable by Adafruit/this core but needs to be looked into by Arduino, seems to be an IDE bug.

closed 2019-12-31 by PTS93