adafruit/ArduinoCore-samd #159: Conflict of ZeroDMA and I2S

FriedrichForstner, 2019-08-01:
Since the addition of ZeroDMA library into the Core we have a conflict with the I2S library's DMA:

`/var/folders/y6/w7n3cq754jv3vkt3vl8tvkkc0000gn/T/arduino_build_504202/libraries/Adafruit_ZeroDMA/Adafruit_ZeroDMA.cpp.o: In function DMAC_Handler:
Adafruit_ZeroDMA.cpp:(.text.DMAC_Handler+0x0): multiple definition of DMAC_Handler
/var/folders/y6/w7n3cq754jv3vkt3vl8tvkkc0000gn/T/arduino_build_504202/libraries/I2S/utility/DMA.cpp.o:DMA.cpp:(.text.DMAC_Handler+0x0): first defined here`

It should be reproducible with the ArduinoSound/AmplitudeSerialPlotter example on a Feather M0 (Arduino 1.8.9, Adafruit ArduinoCore-samd 1.5.0-1.5.2, ArduinoSound 0.2.1, MacOS).

Using DMAC_0_Handler instead (as set for the SAMD51 version) worked/unblock it it for me but I am unsure about other side effects.

ladyada, 2019-08-01:
yeah that library has its own DMA handler... we could either fork/adapt it to use ZeroDMA (so that other libraries can use DMA) or you could try https://github.com/adafruit/Audio

fontanon, 2019-08-05:
Hi @FriedrichForstner, would you please clarify for me how did you used _DMAC_0_Handler_ to make it work? I'd like to give it a try. Thanks in advance.

FriedrichForstner, 2019-08-05:
Hi @fontanon , not sure if it will help you: In Adafruit_ZeroDMA I used the function definition/naming for the samd51 platform (DMAC_0_Handler instead of DMAC_Handler) and thus the naming conflict is resolved. However, I assume that it is also rendering this part non-functional. I am not that familiar with the codebase but please use me for testing.

fontanon, 2019-08-09:
Thanks @FriedrichForstner, I managed to compile after the renaming. However, as you described, weird situations happen with the board: it resets itself constantly.

I'll try another approaches as using the Audio Library fork as @ladyada said:
https://github.com/adafruit/Adafruit_ZeroDMA/issues/12

fontanon, 2019-09-11:
Just to let you know about a functional workaround here: https://github.com/adafruit/Adafruit_ZeroDMA/issues/12

caternuson, 2020-05-01:
Ref:
https://forums.adafruit.com/viewtopic.php?f=57&t=164904

nekuneko, 2020-07-26:
This is related to the Not working on SAMD#5 issue from @arduino-libraries ArduinoLowPower library. 
https://github.com/arduino-libraries/ArduinoLowPower/issues/5