adafruit/ArduinoCore-samd #161: [Question] TCC and TC instances for D21 and D51

CreativeRobotics, 2019-08-05:
I noticed that for D51 devices the TCC and TC instances are listed like this:
TCC0, TCC1, TCC2, TCC3, TCC4, TC0, TC1, TC2, TC3, TC4, TC5
Whereas for the D21 devices they are listed like this:
TCC0, TCC1, TCC2, TC3, TC4, TC5
The datasheets give the D21 TC instances of TC0, TC1 and TC2 ... not TC's 3,4 and 5
I was trying to figure out the reason for this and wondering if it was a mistake, but perhaps one that would break a load of libraries if it were corrected...?

ladyada, 2019-08-05:
no idea, what isnt working?

CreativeRobotics, 2019-08-06:
Sorry - its not that anything is broken, I was trying to teach myself about the inner workings of Arduino board distributions and came across the difference. It seemed anomalous and I was wondering if anyone knew why.

I have since realised that the D21 definitions are inherited from the Arduino CoreSamD repository so its really a question for them.

The only issue it throws up for me is potential confusion for the user - if you use the D21 definitions, for example as they are used in your own zero time library, then the reference in code to TC3 is actually for control of TC module 0 in the datasheet, but if you switch to a D51 chip you are then (correctly?) using TC3 module...

closed 2019-08-06 by CreativeRobotics