adafruit/ArduinoCore-samd #162: samd51 wire hangs when no device connected

GMagician, 2019-08-05:
it seems there is an issue in wire library in function: `bool SERCOM::startTransmissionWIRE(uint8_t address, SercomWireReadWriteFlag flag)`

my AGCM4 boards with marlin configured for an I2C eeprom hangs in:
```c++
    while( !sercom->I2CM.INTFLAG.bit.MB )
    {
      // Wait transmission complete
    }
```

but I don't know why...usually i2c should trasmit and detect missing slave because of ACK missing...how to fix this?

ladyada, 2019-08-05:
you need to have pullups on i2c

GMagician, 2019-08-05:
why are they needed? what happens on bus/Soc if missing (and they are missing since no shield is connected)?

ladyada, 2019-08-05:
because i2c specification requires it. i2c will always hang/crash without them, they are required!

closed 2019-08-05 by GMagician