adafruit/ArduinoCore-samd #163: AGCM4 usb lost written data

GMagician, 2019-08-07:
I'm facing weird issue in Marlin firmware when sending a lot of data consecutively. It seems some bytes are not sent.

I'm looking ad usb code and I see in `USBDeviceClass::send` something like:

```c++
if (usbd.epBank1IsReady(ep)) {
    // previous transfer is still not complete
```
shouldn't be:
```c++
if (!usbd.epBank1IsReady(ep)) {
    // previous transfer is still not complete
```

or ready return false when ready?

GMagician, 2019-08-07:
...seems not the cause

GMagician, 2019-08-07:
ok, found it's using tiny usb...but `serial.write` doesnt check if it can write, it simply skip data...is it normal?

`Adafruit_USBD_CDC::write(uint8_t ch)` skip if buffer full
`Adafruit_USBD_CDC::write(const uint8_t *buffer, size_t size)` retries

hathach, 2019-08-08:
Can you post your simple sketch that reproduce the issue? Also please tell us which board you are testing on.

GMagician, 2019-08-08:
answer I got on PR may answer to your doubts or have I to write code?

hathach, 2019-08-08:
the PR is more than enough, I read this issue before checking out the PR. This is fixed by #164 . Thank you very much

closed 2019-08-08 by hathach

GMagician, 2019-08-08:
@hathach what do you say about my original doubt:
`f (usbd.epBank1IsReady(ep)) {
    // previous transfer is still not complete`

is this test correct? I'm expecting ready return true when ready, not when busy?

hathach, 2019-08-08:
@GMagician I have no idea, since I haven't checked out the Arduino stack. You may want to raise an issue over the original samd core. They are author, they will give a better answer than me.