adafruit/ArduinoCore-samd #166: I2C comms not working with arduino IDE

JDMart04, 2019-08-12:
When trying to use either BNO055 or FXOS8700 IMUs with an M4 Metro, it crashes when executing the example code from sensorapi while using the arduino-ide. The sensors and M4 work fine when using CircuitPython examples. The same wiring was used for both cases and matches that on the Adafruit websites for the devices.

ladyada, 2019-08-12:
hiya please provide some more details including version numbers for IDE, BSPs, and libraries

JDMart04, 2019-08-12:
I am using Arduino ide version 1.8.9
Not sure what you mean by BSPs.
I used latest libraries downloaded this morning.
FXAS21002C - v1.2.0
FXOS8700 - v1.1.3
Unified Sensor - v1.0.3
SAMD Boards - v1.6.12

makermelissa, 2019-08-12:
Hi, I have connected the BNO055 to a Metro M4 and it worked fine without crashing anything with the latest versions. SAMD Boards is your BSP. Please add a post to https://forums.adafruit.com/ since this issue appears to be isolated so we can help you out there. Thanks.

closed 2019-08-12 by makermelissa

JDMart04, 2019-08-12:
I will do as you asked. In which forum should I post my issue?
I have tried using a second, new M4 Metro connected to the BNO055 with the same result - the comm port disconnects after download (when it tries runs the sketch). I also tried a plain Metro and it DOES work as expected.

makermelissa, 2019-08-12:
Probably this one https://forums.adafruit.com/viewforum.php?f=63. Also, if you're using 5V, try switching to 3V.

JDMart04, 2019-08-12:
Thank you. I tried 3V - result was comm port stayed connected but no response from BNO055.

End result, my wires seemed to be too long (~24 inches twisted pairs). I used shorter wires and things worked.