adafruit/ArduinoCore-samd #178: [FR] Intercept bytes coming from serial/usb

GMagician, 2019-09-10:
Is there any possibility to add a "intercept" function in usb/serial handler that can get bytes before they are stored into buffer?
Marlin firmwares requires it to properly handle "emergency parser", currently solution is to completely duplicate usb code.
I see interrupt use a dummy function that can be overriden... maybe it's possible to add a function (initialized to null) that may be called (when initialized) inside class. I may try to add this but before working to go nowhere I would like to know if it may get a change to be merged.
Thanks

GMagician, 2020-04-26:
@ladyada is there any possibility that if I add this option it will be included?

My idea is to have add something like a virtual function
On uart I would like also an "externally" defineable RX and TX buffer size (default to actual behaviour)
This will let me give a better support to Marlin.

please let me know

ladyada, 2020-04-26:
unlikely - you should be able to set up a timer if you need to read data from the serial buffer

GMagician, 2020-04-26:
That's not how it works in marlin. I think I have to duplicate all the core code, that's not the best solution but I see no escape....I'm sad

closed 2020-04-26 by GMagician

ladyada, 2020-04-26:
if you need a larger RX/TX buffer, that's something you can submit a PR, if its a reasonable #ifdef

GMagician, 2020-04-26:
TX & RX is a very easy task to realize:
this code into uart.h
```c++
#ifndef UART_TX_BUFFER_SIZE
  #define UART_TX_BUFFER_SIZE SERIAL_BUFFER_SIZE
#endif
#ifndef UART_RX_BUFFER_SIZE
  #define UART_RX_BUFFER_SIZE SERIAL_BUFFER_SIZE
#endif
private 
  RingBufferN<UART_RX_BUFFER_SIZE> rxBuffer;
  RingBufferN<UART_TX_BUFFER_SIZE> txBuffer;
```

But doing that is senseless without intercepting received chars, then I'm forced to derive uart.cpp and build my own class but...it's impossible without your support.
if I would like to handle this by my self I need to take control of SERCOM0 interrupt handlers but these are reserved by your code. Is there any possibility to mark them as weak into core?

framework does something like
**__attribute__((weak, alias("Dummy_Handler")))**
to let other people to supersede