adafruit/ArduinoCore-samd #191: Unable to cast Adafruit_USBD_CDC to Serial

PTS93, 2019-12-20:
Using 0.7.1 with Adafruit SAMD core 1.5.3 (newest one broken with TinyUSB atm but should be same issue anyways)
I get the following error when trying to use a class that relies on passing an instance of the USB serial interface.
`no known conversion for argument 1 from 'Adafruit_USBD_CDC' to 'Serial_&'`

This works with the default USB core but not when using TinyUSB. I don't really see why though as it seems to be fully compatible and inheriting also from Stream.

You can reproduce the problem with the OSC library with f.e. this example https://github.com/CNMAT/OSC/blob/master/examples/SerialReceive/SerialReceive.ino
and replacing line 9 with `SLIPEncodedUSBSerial SLIPSerial( Serial );`

ladyada, 2019-12-20:
did it ever work with SAMD? 
OSC should try to take in objects that inherit from Stream not HardwareSerial - that way it can handle anything

PTS93, 2019-12-20:
Yes, this works fine with SAMD. Just not in combination with TinyUSB.

ladyada, 2019-12-20:
yeah not sure we can do much about that ... maybe if OSC could use Stream type? that would be best :)