adafruit/ArduinoCore-samd #198: Interface of `Print::availableForWrite()` conflicts with Ethernet library implementation

prototypicalpro, 2019-12-30:
Hello,

It looks like the interface of the function `Print::availableForWrite()` was changed from `virtual int availableForWrite(void)` to `virtual size_t availableForWrite(void)` at some point between v1.5.6 and v1.5.8. This change causes any sketch using the Arduino Ethernet library to fail to compile, as the Arduino Ethernet library [is still using the old interface](https://github.com/arduino-libraries/Ethernet/blob/master/src/Ethernet.h).

I tested this by compiling the [WebClient example sketch](https://github.com/arduino-libraries/Ethernet/blob/master/examples/WebClient/WebClient.ino) for the Feather M0 with Adafruit SAMD core v1.5.8, and received the following error:
```
In file included from C:\Program Files (x86)\Arduino\libraries\Ethernet\examples\WebClient\WebClient.ino:18:0:

C:\Program Files (x86)\Arduino\libraries\Ethernet\src/Ethernet.h:222:14: error: conflicting return type specified for 'virtual int EthernetClient::availableForWrite()'

  virtual int availableForWrite(void);

              ^~~~~~~~~~~~~~~~~

In file included from C:\Users\User\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.8\cores\arduino/Stream.h:26:0,

                 from C:\Users\User\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.8\cores\arduino/HardwareSerial.h:24,

                 from C:\Users\User\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.8\cores\arduino/Arduino.h:78,

                 from C:\Users\User\AppData\Local\Temp\arduino_build_407834\sketch\WebClient.ino.cpp:1:

C:\Users\User\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.8\cores\arduino/Print.h:61:20: error:   overriding 'virtual size_t Print::availableForWrite()'

     virtual size_t availableForWrite() { return 0; }

                    ^~~~~~~~~~~~~~~~~
```

A fix for this issue would be appreciated, as many projects I work on use the [Adafruit Ethernet FeatherWing](https://www.adafruit.com/product/3201), which requires the Arduino Ethernet library to function.

Thanks!

ladyada, 2019-12-30:
yeah there was A Reason to change it but i think everyone else changed theirs back too - we'll be reverting in the next release @hathach

hathach, 2019-12-31:
Very quick report, just found it out yesterday myself :)

PR #199 (2019-12-31): revert 6630fe7 availableForWrite() return int
fix #198
closed 2019-12-31 by hathach