adafruit/ArduinoCore-samd #201: libarm_cortexM4lf_math.a version doesn't match corresponding headers in Arduino

rlcamp, 2020-01-06:
The version of CMSIS used to build variants/feather_m4/libarm_cortexM4lf_math.a seems to be quite a bit older than the version of CMSIS exposed by including <arm_math.h>. If boards.txt is modified as follows:

adafruit_feather_m4.compiler.arm.cmsis.ldflags="-L{runtime.tools.CMSIS-4.5.0.path}/CMSIS/Lib/GCC/" "-L{build.variant.path}" -larm_cortexM4lf_math -mfloat-abi=hard -mfpu=fpv4-sp-d16

then the linker will find the version of libarm_cortexM4lf_math.a matching the include file, but there may be non-obvious side effects to this fix.

ladyada, 2020-01-07:
hmm! good to know, can you submit a PR for this fix (if you could do so for all the boards with the same incorrect line that would be really deeply appreciated) and we'll try it out!

closed 2020-01-07 by rlcamp

ladyada, 2020-01-07:
thank u!